espressif/esp-idf #13391: NimBLE sometimes crashes/fails during connection.

SimonBrix, 2024-03-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I expect the connection to get established, or for the connection to fail "gracefully".

### What is the actual behavior?

I encounter an error, where it fails to discover "services", and after that it seems to start running code in a random place in memory.

### Steps to reproduce.

1. Setup an ESP32-S3 with the NimBLE bleprph example.
2. Setup another ESP32-S3 with a modified version of NimBLE blecent example, which disconnects every time all services are discovered to trigger a new reconnect (this project is attatched): [esp-cent-repeat-pair.zip](https://github.com/espressif/esp-idf/files/14614831/esp-cent-repeat-pair.zip)
3. Make it keep running. In our example it sometimes fails to discover the services, and then it seems to do a "double connect", and then stops. In the actual code of our project, it also leads to it starting to execute random code when this happens, which leads to a crash. It seems to be the same place every time with the same binary, but if the binary is changed, the stack trace is different. (Sometimes it is from the constructer of something which has finished constructing way before the error occurs, in the stack trace from the project below I am not to sure what "0x40375bac: esp_restart_noos at /Users/simonbrix/esp/v5.2.1/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:159" is about.)

### Debug Logs.

```plain
## The debug logs from the esp-cent-repeat-pair example:
W (70983) NimBLE_BLE_CENT: Connection number: 21
I (71984) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

I (72033) NimBLE: disconnect; reason=534 
I (72033) NimBLE: 

I (72033) NimBLE: GAP procedure initiated: discovery; 
I (72035) NimBLE: own_addr_type=0 filter_policy=0 passive=1 limited=0 filter_duplicates=1 
I (72044) NimBLE: duration=forever
I (72048) NimBLE: 

Device name: nimble-bleprph
I (72109) NimBLE: GAP procedure initiated: connect; 
I (72109) NimBLE: peer_addr_type=0 peer_addr=
I (72110) NimBLE: 7c:df:a1:e3:6c:22
I (72115) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=0
I (72128) NimBLE: 

[... 163 lines cut ...]
A10     : 0x00000000  A11     : 0x3fc9e2d0  A12     : 0x00060a20  A13     : 0x00060e23  
A14     : 0x0000ce3d  A15     : 0x3fcb8f8c  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000009  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  
0x400570e8: memset in ROM
0x400570f3: memset in ROM

Backtrace: 0x41fffffe:0x3fcbd800 |<-CORRUPTED

ELF file SHA256: 5684348c6

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375bac
0x40375bac: esp_restart_noos at /Users/simonbrix/esp/v5.2.1/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:159

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2da0
entry 0x403c9914
```

rahult-github, 2025-03-04:
Hi @SimonBrix , 

We have worked on fixing multiple cases to avoid multiple connection / abrupt disconnection which lead to such behaviours ( BLE_GAP_ENC_CHANGE with reason 7 implies no connection exists ) . 

I tried your application on S3 for latest 5.2 release branch and i don't observe any failure for over 100 runs. Suggest to check with latest branch code. 

[success_log.txt](https://github.com/user-attachments/files/19065943/success_log.txt)  

Closing this issue currently, but feel free to reopen in case of any updates / queries.

closed 2025-03-04 by rahult-github

AxelLin, 2025-03-05:
> Hi [@SimonBrix](https://github.com/SimonBrix) ,
>
> We have worked on fixing multiple cases to avoid multiple connection / abrupt disconnection which lead to such behaviours ( BLE_GAP_ENC_CHANGE with [...]

@rahult-github 
The 5.2 branch missed several recent nimble fixes because the submodule is not up to date.
Could you update nimble submodule for esp-idf 5.2 brnach?

rahult-github, 2025-03-05:
Hi @AxelLin , update has already happened internally. Should be visible on github once the sync happens.