espressif/esp-idf #13392: usb host msc example  Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

vwwwvwww, 2024-03-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

\esp\v5.2\esp-idf\examples\peripherals\usb\host\msc\main
我成功执行了usb host msc example,并且运行很成功.
于是我把代码从main.c移动到一个新的task.c文件. 
当按下boot 退出usb app的时候就会报错:

              I (6030) usb_task_main: Done
              I (6050) usb_tsak_main: Deinitializing USB
              Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
              
              Core  1 register dump:
              PC      : 0x42017311  PS      : 0x00060033  A0      : 0x82016671  A1      : 0x3fc95890  
              0x42017311: root_port_callback at :/Espressif/esp/v5.2/esp-idf/components/usb/hub.c:763

但在main.c是正常的.

vwwwvwww, 2024-03-15:
[msc.zip](https://github.com/espressif/esp-idf/files/14616925/msc.zip)
 this  is the example .

peter-marcisovsky, 2025-01-21:
Hi @vwwwvwww sorry for a late reply. 
Is this still a relevant issue?

tore-espressif, 2025-04-07:
This problem should be fixed on latest esp-idf master. 
Feel free to reopen if there are more questions

closed 2025-04-07 by tore-espressif