espressif/esp-idf #13394: s_i2c_send_commands() is unusable because of semaphore mistreatment

sfzhi, 2024-03-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### Espressif SoC revision.

ESP32-C3 revision 0.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Seeed XIAO ESP32C3

### Power Supply used.

USB

### What is the expected behavior?

Functions that call `s_i2c_send_commands(...)` internally, such as `i2c_master_probe(...)`, should perform their intended action.

### What is the actual behavior?

The function times out or hangs indefinitely - depending on the supplied timeout value.

### Steps to reproduce.

1. Call `i2c_new_master_bus(...)`.
2. Call `i2c_master_probe(...)`.

### Debug Logs.

```plain
Nothing is logged.
```

### More Information.

The problem is that `s_i2c_send_commands()` tries to acquire the `i2c_master->cmd_semphr` inside `while()` loop multiple times. That is insane. Only the first time can ever succeed. Subsequent loop iterations will invariably hang or time out. I don't understand how nobody has noticed such an obvious problem.
Besides, the function doesn't release the semaphore consistently before exiting.

mythbuster5, 2024-03-16:
Hello @sfzhi I don't see that is this is a problem. `i2c_master->cmd_semphr` is taken in wile loop indeed for protecting that i2c command will not be re-written before previous cmd is done. So, that semaphore will be given in `s_i2c_write_command` and related functions or in ISR.

Anything wrong when you using the driver?

sfzhi, 2024-03-16:
Hi @mythbuster5, having looked into it a bit deeper, it turns out the problem occurs only when the bus is created with `trans_queue_depth` > 0. In that case the ISR never releases the semaphore, because it behaves differently depending on `i2c_master->asnyc_trans`. I'm guessing then `s_i2c_send_command_async()` should be used instead of `s_i2c_send_commands()`. Right? But `i2c_master_probe()` calls `s_i2c_send_commands()` unconditionally. Is it incompatible with the async mode then?

sfzhi, 2024-03-16:
By the way, I can see that `s_i2c_send_command_async()` is called from the ISR, but it doesn't have `IRAM_ATTR`. Is that intentional?

mythbuster5, 2024-03-16:
@sfzhi right this is put in IRAM when I2C_ISR_IRAM_SAFE is enabled. Right, the compatible with probe and async is not that good... TBH, probe is provided as a helper function compared with legacy. I don't think it's soo widely used.

mythbuster5, 2025-01-21:
This can be closed because the semaphore issue has been solved, thanks for your report!

closed 2025-01-21 by mythbuster5