espressif/esp-idf #13395: i2c_master_probe() does not allow to choose addressing mode

sfzhi, 2024-03-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When calling `i2c_master_probe(...)` it is impossible to specify the addressing mode (7-bit vs. 10-bit). If the function is called immediately after `i2c_new_master_bus(...)` it will apparently use 7-bit mode. If it's called at a later time, it seems to use the addressing mode of the last executed transaction. It should be possible to explicitly specify the addressing mode with a parameter.

Alvin1Zhang, 2024-03-16:
Thanks for reporting.

mythbuster5, 2024-03-16:
I never saw someone uses 10-bits address... So this interface only works with 7 bytes address currently... I will find time to add 10-bits address for this api. Thanks

pggh, 2024-03-16:
And i2c_master_probe does not allow to set the speed either.

mythbuster5, 2024-03-16:
@pggh Yes speed is not a must to be set in probe api. I don't want to make this api too heavy

pggh, 2024-03-16:
@mythbuster5 true but the clock must be low enough the the physical bus. In my case an external temperature sensor with a longer cable is used. The combination of pullup value and cable capacity limits the bus speed.

mythbuster5, 2024-03-16:
@pggh I see.. I thought 100Khz is normal enough for I2C usage..

mythbuster5, 2024-03-18:
@pggh Question. What speed on SCL bus do you suspect in your project?

pggh, 2024-03-18:
@mythbuster5 Because the new driver allows to specify the clock frequency per device, I suggest to enable this for probe too. There is no always correct frequency. A good default could be 10'000 Hz which is slow enough in most cases and a probe would use the bus for about 1 ms.