espressif/esp-idf #13397: i2c speed incorrect after error

pggh, 2024-03-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1 and 5.2.1

### Espressif SoC revision.

ESP32S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

own

### Power Supply used.

External 5V

### What is the expected behavior?

Actual speed on i2c to be as set in i2c_config_t.master.clk_speed

### What is the actual behavior?

I configure i2c master with i2c_param_config and clock speed 10'000 Hz and start the driver with i2c_driver_install.
No device attached to the bus (plug&play scenario).
The first write to the device fails as expected and is using 10'000 Hz clock.
From the second write on the clock is 40'000 Hz (measured with idf 5.1).
For comparison, when doing the same with 100'000 Hz (or 40'000 Hz) the clock on the second write is the same as on the first write, as it should be.

### Steps to reproduce.

Excerpt from the test program:

stop is true, n is 1
```
   const auto l = i2c_cmd_link_create();
   switch (state) {
   case S_READ:
      i2c_master_stop(l);
      break;
   case S_CLOSED:
   case S_STOPPED:
      i2c_master_start(l);
      i2c_master_write_byte(l, (slave_addr << 1) | I2C_MASTER_WRITE, true);
      break;
   default:;
   }
   i2c_master_write(l, buf, n, true);
   if (stop) {
      i2c_master_stop(l);
   }
   const auto stat = i2c_master_cmd_begin(i2c_num, l, pdMS_TO_TICKS(100));
   i2c_cmd_link_delete(l);
   if (stat == ESP_OK) {
      state = stop ? S_STOPPED : S_WRITE;
      printf("wrote:");
      for (unsigned i = 0; i < n; ++i) { printf(" %02X", static_cast<unsigned>(buf[i])); }
      printf(" %s\n", stop ? "stop" : "nostop");
   } else {
      state = S_STOPPED;
      const auto ls = i2c_cmd_link_create();
      i2c_master_stop(ls);
      i2c_master_cmd_begin(i2c_num, ls, pdMS_TO_TICKS(10));
      i2c_cmd_link_delete(ls);
      printf("i2c write failed with %u\n", stat);
      return 2;
   }

   return 0;
```

pggh, 2024-03-21:
When will this be fixed?

closed 2024-03-22 by espressif-bot

KonssnoK, 2024-07-31:
@nebkat is this issue present also on v4.4 ?

nebkat, 2024-07-31:
@KonssnoK I am not a maintainer here - but AFAIK there were no major changes to I2C between 4.4 and 5.1, so it is not impossible that this would also apply to 4.4. You'll have to check the exact source contents to verify.

KonssnoK, 2024-08-03:
@nebkat sorry i saw you committed something and assumed you were involved :D 
thanks anyway, we found the issue we were having