espressif/esp-idf #13398: New IDF 5.2.1 i2c driver does not detect not attached device

pggh, 2024-03-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### Espressif SoC revision.

ESP32S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

own

### Power Supply used.

External 5V

### What is the expected behavior?

Accessing a not attached device causes i2c_master_transmit/i2c_master_receive/i2c_master_transmit_receive to fail.

### What is the actual behavior?

Transfer continues with write/read even when no device is attached to the bus that would ACK the device write address.
The i2c_master_* functions return ES_OK.

### Steps to reproduce.

init:
```
   const i2c_master_bus_config_t bus_conf = {
      .i2c_port = i2c_num,
      .sda_io_num = static_cast<gpio_num_t>(sda),
      .scl_io_num = static_cast<gpio_num_t>(scl),
      .clk_source = I2C_CLK_SRC_DEFAULT,
      .glitch_ignore_cnt = 7,
      .intr_priority = 0,
      .trans_queue_depth = 0,
      .flags = {.enable_internal_pullup = 1},
   };

   if (dev_handle != nullptr) {
      i2c_master_bus_rm_device(dev_handle);
      dev_handle = nullptr;
   }

   if (bus_handle != nullptr) {
      i2c_del_master_bus(bus_handle);
      bus_handle = nullptr;
   }

   if (i2c_new_master_bus(&bus_conf, &bus_handle) != ESP_OK) {
      printf("i2c_new_master_bus failed!\n");
      return 1;
   }
...
   const i2c_device_config_t dev_conf = {
      .dev_addr_length = I2C_ADDR_BIT_LEN_7,
      .device_address = addr,
      .scl_speed_hz = speed,
   };

   if (i2c_master_bus_add_device(bus_handle, &dev_conf, &dev_handle) != ESP_OK) {
      printf("i2c_master_bus_add_device failed!\n");
      i2c_del_master_bus(bus_handle);
      bus_handle = nullptr;
      return false;
   }

```

### More Information.

Because the new driver has so many bugs, I did not investigate the details.

mythbuster5, 2024-03-16:
@pggh Yes, no device connected to bus is allowed now..... Why you want to send data even there is no device?

pggh, 2024-03-16:
@mythbuster5 The device is connected externally. Or it may be not connected, in which case the write should fail.

mythbuster5, 2024-03-16:
@pggh ok, I will set this behavior as a flag let you choose it fails or not.

pggh, 2024-03-16:
@mythbuster5 I can't think of a scenario when a device the does not ACK a write (address or data) should not be detectable.
A failure should always return an error code. (As it used to be in the old driver).
The user/caller can ignore the error if desired. No flag needed, default behavior.
Also the transfer on the bus should end as soon the device noes not ACK a write (address or data). It does not make sense to send any data when the device does not ACK it's address.

mythbuster5, 2024-03-16:
@pggh There indeed a scenario some user asks not return error code in old driver when a devie is not connected to bus. I guess it maybe they only want to say wave on bus? I don't know. You are not the only person ask to return err code when nack is detected. I'm considering to add this back. It's not complicated, only few lines changes. Maybe several days later, it can be on github.

closed 2024-03-20 by espressif-bot