espressif/esp-idf #13399: New IDF 5.2.1 i2c driver appears to use busy wating

pggh, 2024-03-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### Espressif SoC revision.

ESP32S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

own

### Power Supply used.

USB

### What is the expected behavior?

A relatively slow communication device like i2c should never use busy waiting, when when using a synchronous API.

### What is the actual behavior?

In i2c_master.c s_i2c_send_commands I found
'''
    // For blocking implementation, we must wait `done` interrupt to update the status.
    while (i2c_master->trans_done == false) {};
    if (i2c_master->status != I2C_STATUS_ACK_ERROR && i2c_master->status != I2C_STATUS_TIMEOUT) {
        atomic_store(&i2c_master->status, I2C_STATUS_DONE);
    }
'''

This looks like busy-waiting until the transfer completes. I have not analyzed the details, so it could not be that bad as it looks.

In any case, if a task needs to wait for more than a couple of microseconds, it should block and not busy-wait to allow other tasks with same or lower priority to run during the wait time.

### Steps to reproduce.

I had a glance at the sources of the driver while looking into issue #13398.

mythbuster5, 2024-03-16:
@pggh The fix is already merged.... It will sync to github is several days... thanks

suda-morris, 2024-03-18:
Fixed in https://github.com/espressif/esp-idf/commit/1517d61addbc77d6ba33833126e1ec9f91882dfd

mythbuster5, 2024-03-28:
https://github.com/espressif/esp-idf/commit/1517d61addbc77d6ba33833126e1ec9f91882dfd . I will close this issue now, and fix on 5.2 will be sync soon

closed 2024-03-28 by mythbuster5