espressif/esp-idf #13400: New IDF 5.2.1 i2c driver does not allow a transfer to split into cuncks

pggh, 2024-03-16:
### Is your feature request related to a problem?

The new IDF 5.2.1 i2c master driver has a new API that is in many cases easier to use.
However it does not allow to break up a single transfer (start + write_or_read + stop) into several data chunks.

This is not needed very often but there are cases where it is needed. eg. when you do not have all write-data in a buffer because it is streamed from an other communication channel or simply to large to fit in a single buffer.
Similar for read regarding size or if the number of bytes to be read depends on the data read itself.

Whether there is a easy (and performant) workaround depends on the device. As far as I remember some display controller protocols do not simply allow to break up a memory write into multiple start+write+stop transfers because the pixel address where data is written to is reset on every start.

### Describe the solution you'd like.

Add a flag to i2c_master_transmit/i2c_master_receive that determines whether the transfer should be ended (stop) or is allowed to be continued with further calls to i2c_master_transmit/i2c_master_receive. (also a switch from transfer to receive to generate a start without a preceding stop)

Or split the functions into two: i2c_master_transmit_and_stop, i2c_master_transmit_and_continue, same for receive.

In C++ the flag approach with a default argument would be easier to use. For C?

### Describe alternatives you've considered.

The new API simply does not allow this functionality any more. So if the old driver should not be used any more, the new driver needs to be extended.

### Additional context.

In all I2C documentation and all datasheets, the terms 'write' and 'read' are used to express the direction.
Same in the old driver. Why does the new driver use transmit/receive instead?
In IDF 5.2.1 the new driver has many bugs and is probably not widely used.
If you would consider the to change the names to the more common write/read, it's now the last chance without causing too much trouble.

mythbuster5, 2024-03-17:
Why not naming write & read anymore because good name has been occupied by legacy driver. The linker will not tell the difference between them. As for why we call transmit/receive and no support chunks anymore, we refer to CMSIS driver. For one transaction we encourage to prepare whole driver like what we already did in new lcd driver.

For chunks usage. I already consider to have like i2c_job transaction. User can define their job. But that will not be online that soon

pggh, 2024-03-17:
@mythbuster5 regarding naming: my suggestion would be to drop C and use C++ for "new" stuff as it should be done for about 25 years now. You get overloading, namespaces, constants and don't pay any extra for things like this.

matti122, 2024-05-08:
Hi @mythbuster5 
is there any time schedule available when this feature will be added?
I have multiple I2C devices with 8bit and 16bit registers and also need to send 8bit chunks.
Switching to the legacy driver will be no solution for long term.
Could you please advise
Thanks

jrahlf, 2024-10-04:
I second this request.
It is common to have a separation of command and data to be sent to an I2C target.
With the old driver, you could write your own function to have a `master_transmit_transmit` function, but this seems impossible now, since `s_i2c_synchronous_transaction` is static. 
So now, I have to make a temporary copy of command + data into a contiguous array. :(

mythbuster5, 2025-02-18:
This request has been added called `i2c_master_execute_defined_operations`. So I will close this issue. Thanks!

closed 2025-02-18 by mythbuster5