espressif/esp-idf #13401: Wrong I2C_NUM_MAX for ESP32-C6

RDobrinov, 2024-03-17:
https://github.com/espressif/esp-idf/blob/8171b22c40f39a75cd0db62165e7ef017f78c572/components/hal/include/hal/i2c_types.h#L29

I2C_NUM_MAX takes value 2U, but ESP32-C6 has only one I2C port

mythbuster5, 2024-03-17:
Because there is an lp_i2c

RDobrinov, 2024-03-17:
And lp can be used as normal i2c port?
```c
i2c_master_bus_config_t i2c_mst_config = {
    .clk_source = I2C_CLK_SRC_DEFAULT,
    .i2c_port = LP_I2C_NUM_0,
    .scl_io_num = I2C_MASTER_SCL_IO,
    .sda_io_num = I2C_MASTER_SDA_IO,
    .glitch_ignore_cnt = 7,
    .flags.enable_internal_pullup = true,
};

i2c_master_bus_handle_t bus_handle;
ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &bus_handle));
```

mythbuster5, 2024-03-17:
Currently no. This is a feature will be supported in the future. Thanks~!  Currently this is only used for lp_core.

mythbuster5, 2024-03-17:
As for some known new I2C problems have been reported by others will be fixed in several days.~~

AxelLin, 2024-04-09:
> As for some known new I2C problems have been reported by others will be fixed in several days.~~

@mythbuster5 
It's difficult to understand your comment.. what is the issue you mentioned? Is it fixed now?

mythbuster5, 2024-04-22:
This issue is fixed as a new adding feature on master

closed 2024-04-22 by mythbuster5

DexiangYang, 2024-05-27:
https://github.com/espressif/esp-idf/issues/12099#issuecomment-1698574283 @mythbuster5 请问这里提到的问题有解决办法吗，先前的解决方法的代码找不到了，目前看起来如果产生一个NACK后，无法正常访问正常的地址

DexiangYang, 2024-05-27:
检查错误码是ESP_ERR_TIMEOUT错误