espressif/esp-idf #13403: `ble_store_config_cccds` referenced even with CCCD persistence disabled

therealergo, 2024-03-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Esp-Idf to compile with `CONFIG_BT_NIMBLE_MAX_CCCDS=0`

### What is the actual behavior?

Compilation fails, since `ble_store_config_cccds` is referenced even when it is undefined.

### Steps to reproduce.

1. Set `CONFIG_BT_NIMBLE_MAX_CCCDS` to `0` to disable CCCD persistence with Esp-Idf v5.1.2 or newer.
2. Attempt to compile any project.

### Debug Logs.

```plain
ld.exe: esp-idf/bt/libbt.a(ble_store_nvs.c.obj):(.literal.ble_store_config_conf_init+0x8): undefined reference to `ble_store_config_cccds'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed
```

### More Information.

As of commit [b85d99d](https://github.com/espressif/esp-nimble/commit/b85d99de41d8c9085c6f3de50a29ebedd9681381), NimBLE better supports setting `CONFIG_BT_NIMBLE_MAX_CCCDS` to `0` to disable CCCD persistence. This is desirable in some circumstances, e.g. see [this](https://github.com/espressif/esp-idf/issues/12318) previous issue where another person uses this configuration value. When this configuration is set, newer versions of NimBLE do not define the `ble_store_config_cccds` array, since it is not used by the implementation. Unfortunately, the Esp-Idf implementation of CCCD persistence always references the `ble_store_config_cccds` array in `ble_store_config_conf.c` and `ble_store_nvs.c`. As a result, Esp-Idf v5.1.2 will fail to compile if `CONFIG_BT_NIMBLE_MAX_CCCDS` is set to `0`. Checks like `#if MYNEWT_VAL(BLE_STORE_MAX_CCCDS)` should be added around references to this array to ensure that it is not referenced while undefined.

rahult-github, 2024-04-03:
Hi @therealergo , 
 
  I tried the mentioned version at my end and i do not observe the failure you have mentioned.  
  
   Anyways, i have tried to make a fix based on usage. Can you please help check if attached patch works ok for you ?  The patch has to be applied in $IDF_PATH/components/bt/host/nimble/nimble path 
   
   
[fix_cccd_zero_usage.txt](https://github.com/espressif/esp-idf/files/14847171/fix_cccd_zero_usage.txt)

therealergo, 2024-04-05:
Hi @rahult-github

Thanks for looking into this. I suspect you do not see the compilation failure because you have the entire persistence module disabled. You have to set `CONFIG_BT_NIMBLE_NVS_PERSIST` to `y` to enable the persistence feature at all. This includes persistence for bonding data, CCCD's, and encoded advertisement data. I have attached a minimal sdkconfig that exhibits this problem.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/14886154/sdkconfig.txt)

Regardless, the patch fixes the issue. Hope it can be merged in soon!

rahult-github, 2024-04-18:
Change has been [merged](https://github.com/espressif/esp-idf/commit/c8e454b4fac5670bc423027283950af07c771310). Closing

closed 2024-04-18 by rahult-github