espressif/esp-idf #13404: I2C master cannot be shared across Tasks in same Core

matialmar, 2024-03-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Heltec WiFi LoRa v2

### Power Supply used.

USB

### What is the expected behavior?

I should be able to use the i2c_master_bus_handle created in app_main into another task in the same Core.

```c++
...
void app_main(void)
{
    ESP_LOGI(TAG, "Initializing I2C Master");
    //#######################################################################
    //Initialize I2C
    i2c_master_bus_handle_t i2c_bus_handle;

    i2c_master_bus_config_t i2c_bus_config = {
        .clk_source                   = I2C_CLK_SRC_DEFAULT,
        .i2c_port                     = -1,                        //Auto-select
        .scl_io_num                   = I2C_SCL_PIN,
        .sda_io_num                   = I2C_SDA_PIN,
        .glitch_ignore_cnt            = 7,
        .flags.enable_internal_pullup = true ,
    };
    
    esp_err_t err = i2c_new_master_bus(&i2c_bus_config, &i2c_bus_handle);

    ESP_LOGI(TAG, "I2C initialized: %s", esp_err_to_name(err));

    if (err != ESP_OK)
    {
        vTaskSuspend(xTaskGetCurrentTaskHandle());
    }
  
    //-----------------------------------------------------------------------
    // Create BME280 Task
    TaskHandle_t sensor_handle = NULL;

    ESP_LOGI(TAG, "Creating BME280 Task");

    bme280_parameters_t params = {
        .i2cBusHandle = i2c_bus_handle,
    };

    BaseType_t xReturnB = xTaskCreatePinnedToCore(vTaskReadSensor, "BME280", SENSOR_STACK_SIZE, &params, tskIDLE_PRIORITY, &sensor_handle, 0);    

    if( xReturnB == pdFAIL )
    {
        ESP_LOGE(TAG, "Failed to create BME280 task");
    }

    // No need for vTaskStartScheduler in ESP-IDF. So just suspend the task to hold things in memory.
    vTaskSuspend(xTaskGetCurrentTaskHandle());
}
````

Task Read Sensor

```c++
#include "sensor.h"

void vTaskReadSensor(void *pvParameters){

    bme280_parameters_t* params = (bme280_parameters_t*)pvParameters;

    ESP_LOGI(TAG, "Install BME280 driver");

    esp_err_t probe = i2c_master_probe(params->i2cBusHandle, BME280_I2C_ADDR_PRIM, -1);

    ESP_LOGI("BME280", "Probe: %s", probe == ESP_OK ? "OK" : "FAIL");

    if(probe != ESP_OK){
        vTaskSuspend(NULL);
    }
    
    for( ;; )
    {
    }
}
````

### What is the actual behavior?

When i do a i2c_master_probe in the ReadSensor task I get a kernel panic.

### Steps to reproduce.

1. Create an i2c master in app_main()
2. Pass i2c_master_handle as parameter to another task 
3. Run i2c_master_probe in the Task 

### Debug Logs.

```plain
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7172
load:0x40078000,len:15564
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3904
entry 0x40080640
I (29) boot: ESP-IDF v5.3-dev-2486-gc460e1cd7e 2nd stage bootloader
I (29) boot: compile time Mar 17 2024 17:02:41
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v1.0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0aab4h ( 43700) map
I (119) esp_image: segment 1: paddr=0001aadc vaddr=3ffb0000 size=02248h (  8776) load
I (122) esp_image: segment 2: paddr=0001cd2c vaddr=40080000 size=032ech ( 13036) load
I (130) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=16430h ( 91184) map
I (164) esp_image: segment 4: paddr=00036458 vaddr=400832ec size=093b0h ( 37808) load
I (186) boot: Loaded app from partition at offset 0x10000
I (186) boot: Disabling RNG early entropy source...
I (197) cpu_start: Multicore app
I (206) cpu_start: Pro cpu start user code
I (206) cpu_start: cpu freq: 160000000 Hz
I (206) app_init: Application information:
I (209) app_init: Project name:     I2C_BME280
I (214) app_init: App version:      1
I (219) app_init: Compile time:     Mar 17 2024 17:02:43
I (225) app_init: ELF file SHA256:  66217ff4d...
I (230) app_init: ESP-IDF:          v5.3-dev-2486-gc460e1cd7e
I (236) efuse_init: Min chip rev:     v0.0
I (241) efuse_init: Max chip rev:     v3.99 
I (246) efuse_init: Chip rev:         v1.0
I (251) heap_init: Initializing. RAM available for dynamic allocation:
I (258) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (264) heap_init: At 3FFB2B10 len 0002D4F0 (181 KiB): DRAM
I (270) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (277) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (283) heap_init: At 4008C69C len 00013964 (78 KiB): IRAM
I (291) spi_flash: detected chip: gd
I (293) spi_flash: flash io: dio
W (297) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (311) main_task: Started on CPU0
I (321) main_task: Calling app_main()
I (321) MAIN: Initializing I2C Master
I (321) gpio: GPIO[4]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0
I (331) gpio: GPIO[15]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
I (341) MAIN: I2C initialized: ESP_OK
I (341) MAIN: Creating BME280 Task
I (351) BME280 Task: Install BME280 driver

assert failed: xQueueSemaphoreTake queue.c:1713 (pxQueue->uxItemSize == 0)

Backtrace: 0x400817ee:0x3ffb5110 0x400856d5:0x3ffb5130 0x4008b149:0x3ffb5150 0x40085ede:0x3ffb5270 0x400d9a0a:0x3ffb52b0 0x400d5f65:0x3ffb5300 0x400860c5:0x3ffb5320
0x400817ee: panic_abort at /opt/esp/idf/components/esp_system/panic.c:463
0x400856d5: esp_system_abort at /opt/esp/idf/components/esp_system/port/esp_system_chip.c:92
0x4008b149: __assert_func at /opt/esp/idf/components/newlib/assert.c:81
0x40085ede: xQueueSemaphoreTake at /opt/esp/idf/components/freertos/FreeRTOS-Kernel/queue.c:1713 (discriminator 1)
0x400d9a0a: i2c_master_probe at /opt/esp/idf/components/esp_driver_i2c/i2c_master.c:1036
0x400d5f65: vTaskReadSensor at /workspaces/I2C_BME280/main/tasks/sensor.c:156
0x400860c5: vPortTaskWrapper at /opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: 66217ff4d

Rebooting...
```

### More Information.

Using the exact same code but in app_main to read the sensor, everything works fine.

matialmar, 2024-03-18:
Setting trans_queue_depth > 0  avoids this issue but it seems that the i2c master is not being released by the main app even tho no transaction is being explicitly sent.

nopnop2002, 2024-03-18:
&params will pass the address.
This address will be invalidated when app_main exits.
You can prevent invalidation by doing the following:
```
    BaseType_t xReturnB = xTaskCreatePinnedToCore(vTaskReadSensor, "BME280", SENSOR_STACK_SIZE, &params, tskIDLE_PRIORITY, &sensor_handle, 0);    

    if( xReturnB == pdFAIL )
    {
        ESP_LOGE(TAG, "Failed to create BME280 task");
    }
    while(1) { vTaskDelay(1); }
```

matialmar, 2024-03-18:
To the best of my knowledge `vTaskSuspend(xTaskGetCurrentTaskHandle());` suspends the task but does not frees or "invalidate" the address. We would have a "segmentation fault" equivalent if that were the case.

nopnop2002, 2024-03-18:
>To the best of my knowledge vTaskSuspend(xTaskGetCurrentTaskHandle()); suspends the task but does not frees or "invalidate" the address. 

You are right.

nopnop2002, 2024-03-19:
@matialmar 

Try this.
```
#include <stdio.h>
#include <inttypes.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/i2c_master.h"
#include "esp_log.h"

static const char *TAG = "MAIN";

void task(void *pvParameters)
{
    i2c_master_bus_handle_t i2c_bus_handle = (i2c_master_bus_handle_t)pvParameters;
    ESP_LOGI(pcTaskGetName(NULL), "i2c_bus_handle=%p", i2c_bus_handle);
    esp_err_t err = i2c_master_probe(i2c_bus_handle, 0x76, -1);
    ESP_LOGI(TAG, "i2c_master_probe=%d", err);
    vTaskDelete( NULL );
}

#define I2C_SCL_PIN 22
#define I2C_SDA_PIN 21

void app_main()
{
    ESP_LOGI(TAG, "Initializing I2C Master");
    i2c_master_bus_handle_t i2c_bus_handle;

    i2c_master_bus_config_t i2c_bus_config = {
        .clk_source                   = I2C_CLK_SRC_DEFAULT,
        .i2c_port                     = -1,                        //Auto-select
        .scl_io_num                   = I2C_SCL_PIN,
        .sda_io_num                   = I2C_SDA_PIN,
        .glitch_ignore_cnt            = 7,
        .flags.enable_internal_pullup = true ,
    };

    esp_err_t err = i2c_new_master_bus(&i2c_bus_config, &i2c_bus_handle);
    ESP_LOGI(TAG, "i2c_new_master_bus=%d", err);
    ESP_LOGI(TAG, "i2c_bus_handle=%p", i2c_bus_handle);
    err = i2c_master_probe(i2c_bus_handle, 0x76, -1);
    ESP_LOGI(TAG, "i2c_master_probe=%d", err);

    xTaskCreate(&task, "Task1", 2048, i2c_bus_handle, 1, NULL);
    vTaskSuspend(NULL);
}
```

matialmar, 2024-03-19:
@nopnop2002 
That example worked great!  

I tried the same example modifying the `stack_size` and it seems that bellow 2048 wont work. 
Thank you very much for your help!

closed 2024-03-19 by matialmar

andreichalapco, 2024-08-20:
Can we learn something from here? 
xQueueSemaphoreTake at /opt/esp/idf/components/freertos/FreeRTOS-Kernel/queue.c:1713 (discriminator 1) why is assert this not working?

```cpp
    configASSERT( pxQueue->uxItemSize == 0 );
```

Why does an increase in stack size solve this?

matialmar, 2024-08-20:
@andreichalapco Assuming that the Stack Size is the problem, the best guess i can make is that if the stack isn't large enough to hold each value of the structure. And maybe uxItemSize is outside the stack, hence failing the assert?

andreichalapco, 2024-08-20:
@matialmar can it be that another task is writing on the stack where the mutex is located? because there is no loadprohibited error or something like this. I'm currently investigating this because I also encountered the same configAssert error