espressif/esp-idf #13407: New I2c driver and ETH driver used together trigger interrupt WDT timeout

simunma, 2024-03-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### Espressif SoC revision.

ESP32 (revision v3.0)

### Operating System used.

Linux

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

To return from `esp_eth_driver_install(...)` after earlier in code calling `i2c_new_master_bus(...)` from new i2c api.

### What is the actual behavior?

**Interrupt wdt timeout** during executing `esp_eth_driver_install(...)` after calling `i2c_new_master_bus(...)`.

### Steps to reproduce.

1. call `i2c_new_master_bus(...)`
2. call `esp_eth_driver_install(...)`

```
    m_i2c_master_config.clk_source = I2C_CLK_SRC_DEFAULT;
    m_i2c_master_config.i2c_port = I2C_NUM_0;
    m_i2c_master_config.scl_io_num = I2C_SCL_PIN;
    m_i2c_master_config.sda_io_num = I2C_SDA_PIN;
    m_i2c_master_config.glitch_ignore_cnt = 7;
    m_i2c_master_config.flags.enable_internal_pullup = true;
    result = i2c_new_master_bus(&m_i2c_master_config, &m_i2c_naster_bus_handle);

    result = esp_netif_init();
    eth_netif = esp_netif_new(&netif_cfg);

    result = esp_event_loop_create_default();
    emac_config.smi_mdc_gpio_num = BRD_E_ETH_MDC;
    emac_config.smi_mdio_gpio_num = BRD_E_ETH_MDIO;
    mac = esp_eth_mac_new_esp32(&emac_config,&mac_config);

    phy_config.phy_addr = ETH_PHY_ADDRESS;
    phy_config.reset_gpio_num = BRD_E_PHY_RESET;
    phy = esp_eth_phy_new_lan87xx(&phy_config);

    eth_config.mac = mac;
    eth_config.phy = phy;
[... 237 lines cut ...]
D (1103) intr_alloc: Connected src 46 to int 2 (cpu 0)                                                                              
D (1109) spi_flash: trying chip: issi                                                                                               
D (1112) spi_flash: trying chip: gd                                                                                                 
I (1116) spi_flash: detected chip: gd                                                                                               
I (1120) spi_flash: flash io: dio                                                                                                   
D (1124) cpu_start: calling init function: 0x400daf7c                                                                               
D (1129) cpu_start: calling init function: 0x400876d8                                                                               
D (1134) cpu_start: calling init function: 0x400d21c0                                                                               
D (1139) cpu_start: calling init function: 0x400d50f8 on core: 0                                                                    
D (1146) intr_alloc: Connected src 17 to int 3 (cpu 0)                                                                              
D (1151) cpu_start: calling init function: 0x400fc860 on core: 0                                                                    
D (1157) intr_alloc: Connected src 24 to int 9 (cpu 0)                                                                              
D (1162) app_start: Starting scheduler on CPU0                                                                                      
 (1166) intr_alloc: Connected src 25 to int 2 (cpu 1)                                                                               
D (1166) app_start: Starting scheduler on CPU1                                                                                      
;32mI (1166) main_task: Started on CPU0                                                                                             
D (1176) heap_init: New heap initialised at 0x3ffe0440                                                                              
D (1186) heap_init: New heap initialised at 0x3ffe4350                                                                              
D (1186) intr_alloc: Connected src 16 to int 12 (cpu 0)                                                                             
I (1186) main_task: Calling app_main()
```

### More Information.

Legacy I2C API works as expected.

mythbuster5, 2024-03-18:
There will be a fix merged soon~

mythbuster5, 2024-03-28:
There is a fix already merged in master. Could you please have a try on master with step you mentioned above. If there is no more issue. I'm going to conclude this topic,~ @simunma

AxelLin, 2024-04-09:
> There is a fix already merged in master. 

@mythbuster5 
Could you share the commit of the fix? (So people can check if it is fixed in v5.2 branch).

simunma, 2024-04-09:
@mythbuster5 
Hi, sorry i was out of office for holidays.
I tested it today, and the problem seems to be fixed.
The minimal reproducible example and the main project runs as expected using the new API for i2c and ETH driver on current master branch. No more Interrupt wdt timeout.

kostaond, 2024-04-19:
@mythbuster5 thanks for I2C driver fix. @simunma thanks for the testing. Closing the issue...

closed 2024-04-19 by kostaond