espressif/esp-idf #13408: ESP32-C3-DevKitM-1 RS485 pinout

naltamura, 2024-03-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

which pins are used to connect rs-485 on this board? it seems that:
  ------------------------------------------------------------------------------------------------------------------------------
  |  UART Interface       | #define            | Default pin for ESP32 | Default pins for others   | External RS485 Driver Pin |
  | ----------------------|--------------------|-----------------------|---------------------------|---------------------------|
  | Transmit Data (TxD)   | CONFIG_MB_UART_TXD | GPIO23                | GPIO9                     | DI                        |
  | Receive Data (RxD)    | CONFIG_MB_UART_RXD | GPIO22                | GPIO8                     | RO                        |
  | Request To Send (RTS) | CONFIG_MB_UART_RTS | GPIO18                | GPIO10                    | ~RE/DE                    |
  | Ground                | n/a                | GND                   | GND                       | GND                       |
  ------------------------------------------------------------------------------------------------------------------------------
  
  are suggested but on my board there are not GPIO 23 as well as 22.
  GIPO 9 and 8 are the right? 
  
  thanks in advance

nopnop2002, 2024-03-18:
You can change it to any GPIO using uart_set_pin.

https://github.com/espressif/esp-idf/blob/master/examples/peripherals/uart/uart_async_rxtxtasks/main/uart_async_rxtxtasks_main.c

alisitsyn, 2024-03-18:
@naltamura ,

The recommended pins for esp32c3 are in the column `Default pins for others`. If you need to change them you can do it in the menuconfig as suggested above.

Please check the pins available on the [board]:(https://dl.espressif.com/dl/schematics/SCH_ESP32-C3-DEVKITM-1_V1_20200915A.pdf). Also please check the information [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-reference/peripherals/gpio.html) because the GPIO8, GPIO9 are also strapping pins and you need to make sure the level on them is correct during boot.

alisitsyn, 2024-09-18:
The issue will be closed. Feel free to reopen if it is still actual.

closed 2024-09-18 by alisitsyn