espressif/esp-idf #13409: usb_serial_jtag causing watchdog timer reset

MR-VENOM, 2024-03-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-307-g99daa62628

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

Testing the usb_serial_jtag example code on esp32s3

### What is the actual behavior?

Trying to run the usb_serial_jtag code gives 

E (46716) task_wdt: esp_task_wdt_reset(713): task not found

### Steps to reproduce.

1. Craete an example of usb_serial_jtag from idf examples
2. build flash
3. open any terminal and send any bytes.
...
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/14635157/sdkconfig.txt)

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037590c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2da0
entry 0x403c9914
I (26) boot: ESP-IDF v5.2-307-g99daa62628 2nd stage bootloader
I (26) boot: compile time Mar 18 2024 16:06:50
I (26) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
[... 300 lines cut ...]
E (45846) task_wdt: esp_task_wdt_reset(713): task not found
E (45856) task_wdt: esp_task_wdt_reset(713): task not found
E (45866) task_wdt: esp_task_wdt_reset(713): task not found
E (45876) task_wdt: esp_task_wdt_reset(713): task not found
E (45886) task_wdt: esp_task_wdt_reset(713): task not found
E (45896) task_wdt: esp_task_wdt_reset(713): task not found
E (45906) task_wdt: esp_task_wdt_reset(713): task not found
E (45916) task_wdt: esp_task_wdt_reset(713): task not found
E (45926) task_wdt: esp_task_wdt_reset(713): task not found
E (45936) task_wdt: esp_task_wdt_reset(713): task not found
E (45946) task_wdt: esp_task_wdt_reset(713): task not found
E (45956) task_wdt: esp_task_wdt_reset(713): task not found
E (45966) task_wdt: esp_task_wdt_reset(713): task not found
E (45976) task_wdt: esp_task_wdt_reset(713): task not found
E (45986) task_wdt: esp_task_wdt_reset(713): task not found
E (45996) task_wdt: esp_task_wdt_reset(713): task not found
E (46006) task_wdt: esp_task_wdt_reset(713): task not found
E (46016) task_wdt: esp_task_wdt_reset(713): task not found
E (46026) task_wdt: esp_task_wdt_reset(713): task not found
E (46036) task_wdt: esp_task_wdt_reset(713): task not found
E (46046) task_wdt: esp_task_wdt_reset(713): task not found
E (46056) task_wdt: esp_task_wdt_reset(713): task not found
E (46066) task_wdt: esp_task_wdt_reset(713): task not found
E (46076) task_wdt: esp_task_wdt_reset(713): task not found
```

mythbuster5, 2024-03-18:
I feel nervous when I saw this `sleep: Enable automatic switching of GPIO sleep configuration` Because if chip sleeps in any reason. The clock that serial_jtag relies on will be powered off.

MR-VENOM, 2024-03-19:
@mythbuster5 It was not enabled by the program i just took one sample project copied the code from github to main and disabled secondary console in menu config

MR-VENOM, 2024-03-22:
@igrr  any comment on this ?

mythbuster5, 2024-03-27:
What about trying increase `ECHO_TASK_STACK_SIZE` to 4096 in example. It works on my side.. @MR-VENOM

MR-VENOM, 2024-04-06:
@mythbuster5 that solved the issue but there is one more problem that i am facing on esp32h2

`#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/usb_serial_jtag.h"
#include "sdkconfig.h"
#include "esp_log.h"
#include "esp_check.h"

#define BUF_SIZE (1024)
#define ECHO_TASK_STACK_SIZE (4096)

static void echo_task(void *arg)
{
    // Configure USB SERIAL JTAG
    usb_serial_jtag_driver_config_t usb_serial_jtag_config = {
        .rx_buffer_size = BUF_SIZE,
        .tx_buffer_size = BUF_SIZE,
    };

    ESP_ERROR_CHECK(usb_serial_jtag_driver_install(&usb_serial_jtag_config));
    ESP_LOGI("usb_serial_jtag echo", "USB_SERIAL_JTAG init done");

    // Configure a temporary buffer for the incoming data
    uint8_t *data = (uint8_t *) malloc(BUF_SIZE);
    if (data == NULL) {
        ESP_LOGE("usb_serial_jtag echo", "no memory for data");
        return;
    }

    while (1) {

        int len = usb_serial_jtag_read_bytes(data, (BUF_SIZE - 1), 20 / portTICK_PERIOD_MS);

        // Write data back to the USB SERIAL JTAG
        if (len) {
            usb_serial_jtag_write_bytes((const char *) data, len, 20 / portTICK_PERIOD_MS);
            data[len] = '\0';
            ESP_LOG_BUFFER_HEXDUMP("Recv str: ", data, len, ESP_LOG_INFO);
        }
    }
}

void app_main(void)
{
    esp_log_level_set("*",ESP_LOG_NONE);
    xTaskCreate(echo_task, "USB SERIAL JTAG_echo_task", ECHO_TASK_STACK_SIZE, NULL, 10, NULL);
}`

The data is echoing only the first time i connect to the serial 

`---- Closed serial port /dev/ttyACM0 due to disconnection from the machine ----
---- Opened the serial port /dev/ttyACM0 ----
---- Sent utf8 encoded message: "HELLO" ----
HELLO
---- Sent utf8 encoded message: "HELLO" ----`

Here is my sdkconfig:

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/14892629/sdkconfig.txt)

closed 2024-05-21 by MR-VENOM