espressif/esp-idf #13410: [v5.X] TSL1.3 not working because psa_crypto_init is not called

KonssnoK, 2024-03-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

As per title, in branches 5.X of esp-idf TLS 1.3 will fail to connect because the PSA is not initialized.
Following 
https://mbed-tls.readthedocs.io/en/latest/getting_started/psa/

```
Using the PSA Crypto API

If using PSA Crypto, you must initialize the library by calling psa_crypto_init() before any other PSA API.
```

Please let me know if you think `psa_crypto_init` should be called by the user and why.

mahavirj, 2024-03-19:
@KonssnoK 

TLS1.3 support in the protocol examples was added in 5.2 release, please see below:

https://github.com/espressif/esp-idf/blob/6b9c2fca79a7b77a42242d48b0cfa82d612e9b85/components/esp-tls/esp_tls_mbedtls.c#L82-L88

KonssnoK, 2024-03-19:
@mahavirj what you are reporting is not an example but a esp-idf component change.
When are you going to backport to 5.1?
We can't use 5.2 for our development, being 1 month old.

mahavirj, 2024-03-19:
@KonssnoK 

Yes, it is also possible to handle the intialization part in the example. For v5.1, you can cherry-pick the change from https://github.com/espressif/esp-idf/commit/873ac173c45c3e60fe6b4eddedfefefd6db4d8ee in your application code.

cbaechler, 2024-03-19:
Thanks @mahavirj for your response. 

To give more context to this issue, we are trying to connect to a MQTT broker while having TLS1.3 enabled in the settings. I managed to reproduce this issue with the example in `examples/protocols/mqtt/ssl` ([819fbf9](https://github.com/espressif/esp-idf/commit/819fbf9a6876ed52cb49045e1bf9d5e9e6400f57)). 

In menuconfig, I changed the following settings: 

- `CONFIG_MBEDTLS_SSL_PROTO_TLS1_3=y`
- `CONFIG_MBEDTLS_DEBUG=y`

Running this example then results in the following output

```
I (11497) example_common: Connected to example_netif_sta
I (11507) example_common: - IPv4 address: 192.168.125.125,
I (11507) example_common: - IPv6 address: fe80:0000:0000:0000:7edf:a1ff:fee0:7130, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (11517) MQTTS_EXAMPLE: [APP] Free memory: 292384 bytes
I (11527) MQTTS_EXAMPLE: Other event id:7
I (11537) main_task: Returned from app_main()
I (11717) mbedtls: ssl_tls.c:3919 => handshake

I (11717) mbedtls: ssl_msg.c:2358 => flush output

I (11717) mbedtls: ssl_msg.c:2367 <= flush output

I (11717) mbedtls: ssl_tls.c:3838 client state: MBEDTLS_SSL_HELLO_REQUEST

I (11727) mbedtls: ssl_msg.c:2358 => flush output

I (11737) mbedtls: ssl_msg.c:2367 <= flush output

I (11737) mbedtls: ssl_tls.c:3838 client state: MBEDTLS_SSL_CLIENT_HELLO

I (11747) mbedtls: ssl_client.c:924 => write client hello

W (11757) mbedtls: ssl_tls13_generic.c:1568 Perform PSA-based ECDH/FFDH computation.

W (11757) mbedtls: ssl_tls13_generic.c:1606 psa_generate_key() returned -27648 (-0x6c00)

I (11767) mbedtls: ssl_client.c:1012 <= write client hello

I (11777) mbedtls: ssl_tls.c:3930 <= handshake

E (11787) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x6C00
I (11787) esp-tls-mbedtls: Certificate verified.
E (11797) esp-tls: Failed to open new connection
E (11797) transport_base: Failed to open a new connection
E (11807) mqtt_client: Error transport connect
I (11807) MQTTS_EXAMPLE: MQTT_EVENT_ERROR
I (11817) MQTTS_EXAMPLE: Last error code reported from esp-tls: 0x801a
I (11827) MQTTS_EXAMPLE: Last tls stack error number: 0x6c00
I (11827) MQTTS_EXAMPLE: Last captured errno : 0 (Success)
I (11837) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED
```

Debugging this shows that apparently `psa_get_empty_key_slot` returns `PSA_ERROR_BAD_STATE`. This hints me towards that `psa_crypto_init` is not called. 

Let me try to cherry-pick the change as you suggest and check whether it fixes our problem.

KonssnoK, 2024-03-28:
v5.1 support to be added by espressif. @mahavirj please tell us when.

Our issues with TLS 1.3 continue also on v5.2 
https://github.com/espressif/esp-idf/issues/13489

mahavirj, 2024-03-29:
@KonssnoK 

As I understand from the discussion above, your application is based on the mbedTLS APIs and hence not dependent on the `esp-tls` component. In that case, supporting TLS1.3 is purely application level change and you may refer to my previous comment https://github.com/espressif/esp-idf/issues/13410#issuecomment-2006483086 for the change here. If you have any other specific requirement in v5.1 release, then we can consider the backport as well. Let us know.

KonssnoK, 2024-03-29:
> @KonssnoK
> 
> As I understand from the discussion above, your application is based on the mbedTLS APIs and hence not dependent on the `esp-tls` component. In that case, supporting TLS1.3 is purely application level change and you may refer to my previous comment [#13410 (comment)](https://github.com/espressif/esp-idf/issues/13410#issuecomment-2006483086) for the change here. If you have any other specific requirement in v5.1 release, then we can consider the backport as well. Let us know.

@mahavirj 
no, our application is based exclusively on esp-idf components : mqtt + esp-tls + esp-wifi.

The point of this ticket is that version 5.1, which is our currently selected branch for next release (being more stable than 5.2), does not implement this change https://github.com/espressif/esp-idf/commit/873ac173c45c3e60fe6b4eddedfefefd6db4d8ee and therefore makes TLS1.3 unavailable on esp-idf 5.1

cbaechler, 2024-04-02:
Cherry-picking this commit https://github.com/espressif/esp-idf/commit/7fd1378fbb0b81231a83f91f8227f8fb083635a5 on `release/v5.1` adds the missing call to `psa_crypto_init` and TLS v1.3 seems to work. The commit affects exclusively `components/esp-tls/esp_tls_mbedtls.c` and is present on `master` and `release/v5.2` branches, but not on `release/v5.1`. 

However I am not sure if there are other things missing or cherry-picking this isolated commit would be enough.

mahavirj, 2024-04-02:
@cbaechler

We have considered TLS1.3 as new feature and following are few commits that went in recently:

7fd1378fbb0b81231a83f91f8227f8fb083635a5
873ac173c45c3e60fe6b4eddedfefefd6db4d8ee
dd6229e5cb08289c0f6e6d71cc52c59d50f546ec
11715c5caf4a2e4aedbf3c04eaf0a8760eeedbd6
27681a507373522f0a84fa0478eeab4921d1930b
2da0b61374b30c0d1ecae6ffd90c6ea4361ccdf5

They are isolated enough and should be possible to be cherry-picked for v5.1 (as per the requirement). I am not sure if we can backport all these changes for v5.1, but we will discuss and evaluate once.

KonssnoK, 2024-04-02:
> @cbaechler
> 
> We have considered TLS1.3 as new feature and following are few commits that went in recently:
> 
> [7fd1378](https://github.com/espressif/esp-idf/commit/7fd1378fbb0b81231a83f91f8227f8fb083635a5) [873ac17](https://github.com/espressif/esp-idf/commit/873ac173c45c3e60fe6b4eddedfefefd6db4d8ee) [dd6229e](https://github.com/espressif/esp-idf/commit/dd6229e5cb08289c0f6e6d71cc52c59d50f546ec) [11715c5](https://github.com/espressif/esp-idf/commit/11715c5caf4a2e4aedbf3c04eaf0a8760eeedbd6) [27681a5](https://github.com/espressif/esp-idf/commit/27681a507373522f0a84fa0478eeab4921d1930b) [2da0b61](https://github.com/espressif/esp-idf/commit/2da0b61374b30c0d1ecae6ffd90c6ea4361ccdf5)
> 
> They are isolated enough and should be possible to be cherry-picked for v5.1 (as per the requirement). I am not sure if we can backport all these changes for v5.1, but we will discuss and evaluate once.

@mahavirj as far as i know TLS 1.3 is selectable in 5.1, meaning the feature should be supported (unless that is an error)

hmalpani, 2024-05-13:
Hello @KonssnoK 
We have backported the commits to previous release branches. They should be available soon on GitHub.

KonssnoK, 2024-05-13:
cool, thanks for letting us know

hmalpani, 2024-05-24:
Hello @KonssnoK 
The backported commits are now available on [release/v5.0](https://github.com/espressif/esp-idf/tree/release/v5.0) and [release/v5.1](https://github.com/espressif/esp-idf/tree/release/v5.1) branches on GitHub.

closed 2024-05-24 by hmalpani