espressif/esp-idf #13413: Building project using idf.py build on Windows

krupis, 2024-03-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0.3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Being able to build the project using the ```idf.py build``` via the cmd or powershell

### What is the actual behavior?

I have already discovered that on Windows, if you want to use any of the tools you must use full python and full tool path.
For example, simply using ```idf.py build``` will not work. So I would need to use the following command instead:
```
C:\Users\petrikas.lu\.espressif5\python_env\idf5.0_py3.11_env\Scripts\python.exe C:\Users\petrikas.lu\esp5_0_4\esp-idf\tools\idf.py build
```

However, when trying to execute this command, I get the following error:

```
PS C:\Users\petrikas.lu\Desktop\WORK\systemair\systemair2\systemair-fw> C:\Users\petrikas.lu\.espressif5\python_env\idf5.0_py3.11_env\Scripts\python.exe C:\Users\petrikas.lu\esp5\esp-idf\tools\idf.py build    
C:\Users\petrikas.lu\esp5\esp-idf\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: all (aliases: build)
ESP-IDF v5.0.3
"cmake" must be available on the PATH to use idf.py
```

So I went ahead and installed cmake and added it to my PATH. Then I tried to build the project via cmd again and getting the following error:
```
PS C:\Users\petrikas.lu\Desktop\WORK\systemair\systemair2\systemair-fw> C:\Users\petrikas.lu\.espressif5\python_env\idf5.0_py3.11_env\Scripts\python.exe C:\Users\petrikas.lu\esp5\esp-idf\tools\idf.py build
C:\Users\petrikas.lu\esp5\esp-idf\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: all (aliases: build)
Running cmake in directory C:\Users\petrikas.lu\Desktop\WORK\systemair\systemair2\systemair-fw\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 C:\Users\petrikas.lu\Desktop\WORK\systemair\systemair2\systemair-fw"...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: xtensa-esp32-elf-gcc
CMake Error at C:/Users/petrikas.lu/esp5/esp-idf/tools/cmake/project.cmake:426 (__project):
  The CMAKE_C_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.  Perhaps the extension is
  missing?

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:21 (project)

CMake Error at C:/Users/petrikas.lu/esp5/esp-idf/tools/cmake/project.cmake:426 (__project):
  The CMAKE_CXX_COMPILER:

    xtensa-esp32-elf-g++

  is not a full path and was not found in the PATH.  Perhaps the extension is
  missing?

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:21 (project)

CMake Error at C:/Users/petrikas.lu/esp5/esp-idf/tools/cmake/project.cmake:426 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.  Perhaps the extension is
  missing?

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:21 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
HINT: Try to reinstall the toolchain for the chip that you trying to use. 
For more information run 'idf.py docs -sp get-started/#installation' and follow the instructions for your system
cmake failed with exit code 1, output of the command is in the C:\Users\petrikas.lu\Desktop\WORK\systemair\systemair2\systemair-fw\build\log\idf_py_stderr_output_25616 and C:\Users\petrikas.lu\Desktop\WORK\systemair\systemair2\systemair-fw\build\log\idf_py_stdout_output_25616
```

Could someone clarify to me what could be this issue and why it happens? Why is it so complicated to build the esp-idf projects on Windows via the cmd? 

Any help is appreciated

### Steps to reproduce.

1. Use Windows machine
2. Install esp-idf via the vscode extension
3. Try to build the project using idf.py build via cmd or powerhsell

dobairoland, 2024-03-19:
> I have already discovered that on Windows, if you want to use any of the tools you must use full python and full tool path.

No that is usually not needed. You can see if you find a workaround for one issue then another appears. This is because you your environment was not properly set up.  Did you followed the setup guide at https://github.com/espressif/vscode-esp-idf-extension? Please try to clean up your system and got through it again.

krupis, 2024-03-19:
> No that is usually not needed. You can see if you find a workaround for one issue then another appears. This is because you your environment was not properly set up. Did you followed the setup guide [...]

I have made a fresh v5.2.1 esp-idf install using the the EXPRESS option and after the installation it seems that I can use ```idf.py build``` command.

I am not sure what could have gone wrong with my previous esp-idf version installation because I believe I have also installed it using the same method, I must have messed it up somehow.

Anyways, thanks for clarifying.

mbratch, 2024-03-19:
> I am not sure what could have gone wrong with my previous esp-idf version installation because I believe I have also installed it using the same method, I must have messed it up somehow.

In hindsight, it seems that somehow your environment variables, particular PATH, were not configured properly in the prior setup. The ESP-IDF installer for Windows will update these, but after installation, I always manually check my environment variables just to be sure!  Sometimes upgrading to a newer ESP-IDF I have to fix something in the environment variables.