espressif/esp-idf #13414: esp_mqtt_client_subscribe macro _Generic error for const char* topic type

azhel12, 2024-03-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Expected that `esp_mqtt_client_subscribe` with `const char*` second param will be expanded to `esp_mqtt_client_subscribe_single`

### What is the actual behavior?

Build failed because `esp_mqtt_client_subscribe` macro does not support `const char*`, only `char*`

### Steps to reproduce.

1. Call esp_mqtt_client_subscribe **from C source**, pass `const char*` as second arg.
2. Try build

### Build or installation Logs.

```plain
/root/esp/v5.2.1/esp-idf/components/mqtt/esp-mqtt/include/mqtt_client.h:464:84: error: '_Generic' selector of type 'const char *' is not compatible with any association
```

### More Information.

API Reference instruct use `char*` if I want to single subscribe, but in fact inside esp-idf does not change this code and it should be const. Cast to `char*` seems bad practice. Besides `esp_mqtt_client_subscribe_single` expect `const char*` param, so it is types mismatch. You can just add `const char *: esp_mqtt_client_subscribe_single, \ ` in 465 line of mqtt_client.h.

KaeLL, 2024-03-19:
https://github.com/espressif/esp-mqtt/commit/32102558d3c0d58c8ed45e0029a29da26cb398a5#r139737900

KaeLL, 2024-03-19:
@euripedesrocha

azhel12, 2024-03-20:
> [espressif/esp-mqtt@3210255#r139737900](https://github.com/espressif/esp-mqtt/commit/32102558d3c0d58c8ed45e0029a29da26cb398a5#r139737900)

Yep:)

closed 2024-03-20 by azhel12

Zeni241, 2024-04-07:
I am still having this error in idf-5.2.1.

KaeLL, 2024-04-08:
This is an bug, not a feature. It's been fixed on master, and should be backported to release/v5.2 at least.

euripedesrocha, 2024-04-09:
Hi @KaeLL and @Zeni241 this will be backported.

Zeni241, 2024-05-24:
I found that its still not be backported. Any hope in near future????

Zeni241, 2024-06-04:
@euripedesrocha Thanks for info, I will check and report if any problem.