espressif/esp-idf #13415: Not able to flash programs to esp32c6 getting "A fatal error occurred: Failed to connect to ESP32-C6: Invalid head of packet (0x66): Possible serial noise or corruption." error

arunvarghese12, 2024-03-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I was using ESP32-C6-MINI-1U with the ESP Prog.
made connections as per below image, as suggested by the document:  https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/current-consumption-measurement-modules.html#

[image]

But on  issuing flash, I'm getting "A fatal error occurred: Failed to connect to ESP32-C6: Invalid head of packet (0x66): Possible serial noise or corruption." error and am not able to flash the programs to the device.

on issuing monitor getting the below response continuously,

invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff

Please help resolve this and program the module.

dobairoland, 2024-03-19:
Do you experience the same issue without Joulescope connected? 

What esptool version are you using?

Please add the `--trace` argument to the esptool flashing command and share the output.

arunvarghese12, 2024-03-19:
Hi,
I'm not using joulescope but directly connecting to a power supply.
I'm not using esptool but ESP IDF 5.2

dobairoland, 2024-03-19:
I see only 5 cables going to the module. What is missing?

You can invoke esptool directly. You see the command in the output of `idf.py flash`.

arunvarghese12, 2024-03-19:
6 cables are going, TX, RX, enable, IO0, 3V3 and GND.

I didn't really get what you meant by invoking esptool.

below are the responses I'm getting for --trace command, can you please provide the correct format,

[image]

dobairoland, 2024-03-19:
```
cd build
esptool.py --chip esp32c6 --trace write_flash @flash_args
```

And please send the respose as text file and not as screenshot.

arunvarghese12, 2024-03-19:
this is what I'm getting when giving esptool.py --chip esp32c6 --trace write_flash 
esptool write_flash: error: the following arguments are required: <address> <filename>
what should be @flash_args ?

also the esptool version is, esptool.py v4.7.0

dobairoland, 2024-03-19:
> what should be @flash_args

It should be passed as is. It would take the arguments from that file.

It doesn't matter. From the output it is clear that reset works but it won't put the chip into download mode. There is some issue with the Boot connection. It might help to hold the BOOT button when starting `idf.py flash`.

> IO0

The Boot pin for ESP32-C6 is IO9 if I'm not mistaken.

arunvarghese12, 2024-03-19:
I've changed boot pin to 1O9, still seeing same error, "A fatal error occurred: Failed to connect to ESP32-C6: Invalid head of packet (0x69): Possible serial noise or corruption."

below is the response I'm getting on giving "esptool.py --chip esp32c6 --trace write_flash @flash_args " command.

PS C:\Espressif\frameworks\esp-idf-v5.2.1-2\examples\system\deep_sleep> cd build
PS C:\Espressif\frameworks\esp-idf-v5.2.1-2\examples\system\deep_sleep\build> esptool.py --chip esp32c6 --trace write_flash @flash_args
usage: esptool write_flash [-h] [--erase-all] [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}] [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}] [--spi-connection SPI_CONNECTION] [--no-progress] [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]] [--ignore-flash-encryption-efuse-setting] [--force] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: the following arguments are required: <address> <filename>

arunvarghese12, 2024-03-20:
Hi @dobairoland,
any update on this?

dobairoland, 2024-03-20:
Yep, sorry. On Windows you need apostrophes as well, i.e. `"@flash_args"`. But this is not causing your problem.

If you insist on flashing modules separately then you can experience such issues. Development boards have additional circuitry to ensure that the chip can be put automatically into download mode. This is obviously not working for you. You need to put the chip manually into download mode. You can read more about this at: https://docs.espressif.com/projects/esptool/en/latest/esp32c6/advanced-topics/boot-mode-selection.html#manual-bootloader (the whole page might be useful for you). As I pointed out earlier, this can be achieved by holding the BOOT button while running the flashing command.

I'm closing this issue because I don't think anything can be done on the software side.

closed 2024-03-20 by dobairoland

arunvarghese12, 2024-03-20:
I tried programming on the development board, but on j5, the current measurement rail. On deep sleep I'm measuring 70uA whereas the datasheet number is 7uA, that's why I moved the evaluation to the module, also the EVK doesn't have an external antenna to evaluate RF parameters of esp32-c6.

I've tries holding BOOT button while programming the device but was facing the same error as "invalid header or packet". 

anyway thanks for your response, please update if there is any further debug to be done.

Thanks Again

dobairoland, 2024-03-20:
You can try to pull down GPIO9 by connecting it to GND. You don't have to use the BOOT button.

GPIO8 must be pulled up (https://docs.espressif.com/projects/esptool/en/latest/esp32c6/advanced-topics/boot-mode-selection.html#gpio8).

arunvarghese12, 2024-03-20:
I have pulled down IO9 by connecting it to GND and have connected IO8 to BOOT,

After that on flashing, below is the response I'm getting, the device is not going to the intended state,

C:\Espressif\frameworks\esp-idf-v5.1.2\examples\system\deep_sleep>idf.py -p COM6 flash
Executing action: flash
Running ninja in directory C:\Espressif\frameworks\esp-idf-v5.1.2\examples\system\deep_sleep\build
Executing "ninja flash"...
[1/5] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v5.1.2\examples\system\deep_sleep\build\esp-idf\esptool_py && C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v5.1.2/components/partition_table/check_sizes.py --offset 0x8000 partition --type app C:/Espressif/frameworks/esp-idf-v5.1.2/examples/system/deep_sleep/build/partition_table/partition-table.bin C:/Espressif/frameworks/esp-idf-v5.1.2/examples/system/deep_sleep/build/deep_sleep.bin"
deep_sleep.bin binary size 0x23400 bytes. Smallest app partition is 0x100000 bytes. 0xdcc00 bytes (86%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v5.1.2\examples\system\deep_sleep\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v5.1.2/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Espressif/frameworks/esp-idf-v5.1.2/examples/system/deep_sleep/build/bootloader/bootloader.bin"
Bootloader binary size 0x55c0 bytes. 0x2a40 bytes (33%) free.
[2/3] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v5.1.2\components\esptool_py && C:\Espressif\tools\cmake\3.24.0\bin\cmake.exe -D IDF_PATH=C:/Espressif/frameworks/esp-idf-v5.1.2 -D SERIAL_TOOL=C:/Espressif/python_env/idf5.1_py3.11_env/Scripts/python.exe;;C:/Espressif/frameworks/esp-idf-v5.1.2/components/esptool_py/esptool/esptool.py;--chip;esp32c6 -D SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args -D WORKING_DIRECTORY=C:/Espressif/frameworks/esp-idf-v5.1.2/examples/system/deep_sleep/build -P C:/Espressif/frameworks/esp-idf-v5.1.2/components/esptool_py/run_serial_tool.cmake"
esptool.py --chip esp32c6 -p COM6 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 deep_sleep.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.7.dev3
Serial port COM6
Connecting....
Chip is ESP32-C6FH4 (QFN32) (revision v0.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: 40:4c:ca:ff:fe:48:58:78
BASE MAC: 40:4c:ca:48:58:78
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00033fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 21952 bytes to 13251...
Writing at 0x00000000... (100 %)
Wrote 21952 bytes (13251 compressed) at 0x00000000 in 0.6 seconds (effective 288.8 kbit/s)...
Hash of data verified.
Compressed 144384 bytes to 75149...
Writing at 0x00010000... (20 %)
Writing at 0x00019b81... (40 %)
Writing at 0x000204a4... (60 %)
Writing at 0x000275a6... (80 %)
Writing at 0x0002efb0... (100 %)
Wrote 144384 bytes (75149 compressed) at 0x00010000 in 2.2 seconds (effective 523.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 509.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done

dobairoland, 2024-03-20:
> have connected IO8 to BOOT,

No, that has to be connected to VCC, not BOOT.

But since you flashed once successfuly, now you can do your measurements with deep sleep.

dobairoland, 2024-03-20:
This is from the datasheet of C6:

[image]

https://www.espressif.com/sites/default/files/documentation/esp32-c6_datasheet_en.pdf

arunvarghese12, 2024-03-20:
> But since you flashed once successfuly, now you can do your measurements with deep sleep.

even though I have the previous shared response, the device is not going to sleep,

on giving monitor, below response is what I'm getting,

Enabling timer wakeup, 20s
I (269) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
Enabling GPIO wake����ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x4 (DOWNLOAD(USB/UART0/SDIO_FEI_FEO))
waiting for download

now if I press the BOOT button, it only gives continues log, but still not going to deep sleep

dobairoland, 2024-03-20:
This is download mode. This is needed for flashing. See "waiting for download".

You need normal operation mode to application to be executed from flash (SPI boot).

arunvarghese12, 2024-03-20:
so after flash, I should keep I08 floating and 109 at boot pin?

arunvarghese12, 2024-03-22:
Hi @dobairoland,

I can only flash the device in Download mode but then it is not going to deep sleep state.

How can I configure the module in different states like Deep Sleep?

Regards,
Arun

dobairoland, 2024-03-25:
> so after flash, I should keep I08 floating and 109 at boot pin?

Not floating. It can be any value according to the table. Not boot pin but 1 (pull-up).

> I can only flash the device in Download mode but then it is not going to deep sleep state.

After flashing you need SPI Boot mode to boot into the application which will put it into deep sleep state.

arunvarghese12, 2024-03-25:
> After flashing you need SPI Boot mode to boot into the application which will put it into deep sleep state.

Is this possible in my case, If yes how?

I tried to move the device from download mode to boot mode by pulling up GPIO9, But then the device is going into some other mode, reading current in nanoAmpheres and then rising it gradually.

arunvarghese12, 2024-03-29:
Hi,

> If you insist on flashing modules separately then you can experience such issues. Development boards have additional circuitry to ensure that the chip can be put automatically into download mode.

as you suggested, I've flashed the image on the development board, but on configuring deep sleep, I'm measuring 53uA current on the dev kit but according to data sheet the expected Deep Sleep current is 7uA.
Like that on light sleep, I'm measuring 360uA whereas data sheet number is 180uA.