espressif/esp-idf #13416: Bluetooth reconnection still requires pairing

jjj19950710, 2024-03-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32

### Power Supply used.

USB

### What is the expected behavior?

Bluetooth reconnection does not require pairing。

### What is the actual behavior?

When I reset the board using the esp_restart() function, and then start Bluetooth to reconnect the phone, I need to re-pair. However, after resetting the board and enabling Bluetooth, the phone does not need to be re-paired to reconnect to the development board.

### Steps to reproduce.

1. Modified a2dp_sink examples。
```
void start_run(void)
{
    /* initialize NVS — it is used to store PHY calibration data */
    // esp_err_t err = nvs_flash_init();
    // if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
    //     ESP_ERROR_CHECK(nvs_flash_erase());
    //     err = nvs_flash_init();
    // }
    // ESP_ERROR_CHECK(err);
    esp_err_t err;
    /*
     * This example only uses the functions of Classical Bluetooth.
     * So release the controller memory for Bluetooth Low Energy.
     */
    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_BLE));

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    if ((err = esp_bt_controller_init(&bt_cfg)) != ESP_OK) {
        ESP_LOGE(BT_AV_TAG, "%s initialize controller failed: %s\n", __func__, esp_err_to_name(err));
        return;
    }
    if ((err = esp_bt_controller_enable(ESP_BT_MODE_CLASSIC_BT)) != ESP_OK) {
        ESP_LOGE(BT_AV_TAG, "%s enable controller failed: %s\n", __func__, esp_err_to_name(err));
        return;
    }
    if ((err = esp_bluedroid_init()) != ESP_OK) {
        ESP_LOGE(BT_AV_TAG, "%s initialize bluedroid failed: %s\n", __func__, esp_err_to_name(err));
        return;
    }
    if ((err = esp_bluedroid_enable()) != ESP_OK) {
        ESP_LOGE(BT_AV_TAG, "%s enable bluedroid failed: %s\n", __func__, esp_err_to_name(err));
        return;
    }

#if (CONFIG_BT_SSP_ENABLED == true)
    /* set default parameters for Secure Simple Pairing */
    esp_bt_sp_param_t param_type = ESP_BT_SP_IOCAP_MODE;
    esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_IO;
    esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));
#endif

    /* set default parameters for Legacy Pairing (use fixed pin code 1234) */
    esp_bt_pin_type_t pin_type = ESP_BT_PIN_TYPE_FIXED;
    esp_bt_pin_code_t pin_code;
    pin_code[0] = '1';
    pin_code[1] = '2';
    pin_code[2] = '3';
    pin_code[3] = '4';
    esp_bt_gap_set_pin(pin_type, 4, pin_code);

    bt_app_task_start_up();
    /* bluetooth device name, connection mode and profile set up */
    bt_app_work_dispatch(bt_av_hdl_stack_evt, BT_APP_EVT_STACK_UP, NULL, 0, NULL);
}

void app_main(void)
{
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
    while (1)
    {
        int opcode = 0;

        char input[128] = "\0";

        gets(input);

        if(memcmp("con", input, 3) == 0){
            opcode = 1; //start pair
        }
        else if(memcmp("reset", input, 5) == 0){//2  avdtp 
            opcode = 2; //reset
        }
        switch (opcode)
         {
                case 1:
                    if(input[3] == 48){
                        scan_flag = 0;//wait connect
                    }
                    else if(input[3] == 49){
                        scan_flag = 1;//active connect
                    }
                    vTaskDelay(100/portTICK_PERIOD_MS);
                    memcpy(remote_device_name,input + 4 , sizeof(input)-4);
                    start_run();
                    break;
                case 2:
                    esp_restart();
                    break;
                default:
                     break;
         }
        vTaskDelay(4);
    }
}
```
static void bt_app_gap_cb(esp_bt_gap_cb_event_t event, esp_bt_gap_cb_param_t *param)
```
case ESP_BT_GAP_DISC_STATE_CHANGED_EVT: {
        if (param->disc_st_chg.state == ESP_BT_GAP_DISCOVERY_STOPPED) {
            if (s_a2d_state == APP_AV_STATE_DISCOVERED) {
                s_a2d_state = APP_AV_STATE_CONNECTING;
                ESP_LOGI(BT_AV_TAG, "Device discovery stopped.");
                ESP_LOGI(BT_AV_TAG, "a2dp connecting to peer: %s", s_peer_bdname);
                /* connect source to peer device specificed by Bluetooth Device Address */
                esp_a2d_sink_connect(s_peer_bda);    //add sink_connect
            } else {
                /* not discovered, continue to discover */
                ESP_LOGI(BT_AV_TAG, "Device discovery failed, continue to discover...");
                //esp_bt_gap_start_discovery(ESP_BT_INQ_MODE_GENERAL_INQUIRY, 10, 0);
            }
        } else if (param->disc_st_chg.state == ESP_BT_GAP_DISCOVERY_STARTED) {
            ESP_LOGI(BT_AV_TAG, "Discovery started.");
        }
        break;
    }
```
3. idf.py -p COM3 flash monitor
4. run:   
```
con1HONOR   //active connect
reset     //Reset ESP32
con1HONOR    //active connect
```
The second connection requires pairing

```
con0HONOR   //wait connect
reset     //Reset ESP32
con0HONOR    //wait connect
```
The second connection does not require pairing

Why is this happening?

jjj19950710, 2024-03-19:
the main.c is
[main.zip](https://github.com/espressif/esp-idf/files/14649716/main.zip)

shangke1112, 2024-03-25:
Please upload  the two different  logs.

jjj19950710, 2024-03-26:
con1HONOR   log:
```
I (4505) BT_BTM: btm_sec_set_security_level : sec: 0x0

I (4505) BT_BTM: BTM_SEC_REG[0]: id 0, is_orig 0, psm 0x0001, proto_id 0, chan_id 0

I (4505) BT_BTM:                : sec: 0x80, service name [Service Discovery] (up to 21 chars saved)

I (4515) BT_BTM: btm_sec_set_security_level : sec: 0x0

I (4525) BT_BTM: BTM_SEC_REG[0]: id 0, is_orig 1, psm 0x0001, proto_id 0, chan_id 0

I (4535) BT_BTM:                : sec: 0x80, service name [Service Discovery] (up to 21 chars saved)

I (4545) BT_BTM: btm_sec_set_security_level : sec: 0x0

I (4545) BT_BTM: BTM_SEC_REG[1]: id 50, is_orig 1, psm 0x001f, proto_id 0, chan_id 0

I (4555) BT_BTM:                : sec: 0x80, service name [] (up to 21 chars saved)

I (4565) BT_BTM: btm_sec_set_security_level : sec: 0x0

I (4575) BT_BTM: BTM_SEC_REG[1]: id 50, is_orig 0, psm 0x001f, proto_id 0, chan_id 0

I (4585) BT_BTM:                : sec: 0x80, service name [] (up to 21 chars saved)

I (4725) BT_BTM: BTM_SetPinType: pin type 0 [variable-0, fixed-1], code , length 0

I (4725) BT_BTM: btm_sec_set_security_level : sec: 0x0

I (4735) BT_BTM: BTM_SEC_REG[2]: id 42, is_orig 0, psm 0x0003, proto_id 3, chan_id 0

I (4735) BT_BTM:                : sec: 0x80, service name [RFC_MUX
] (up to 21 chars saved)

I (4745) BT_BTM: BTM_SetInquiryMode

I (4755) BT_BTM: BTM_SetPageScanType

I (4755) BT_BTM: BTM_SetInquiryScanType

I (4775) BT_BTM: BTM_SecAddDevice, link key type:5

I (4775) BT_BTM: BTM_SecAddDevice, link key type:5

I (4775) BT_BTM: BTM_SecAddDevice, link key type:4

I (4785) BT_BTM: BTM_InqDbRead: bd addr [6eb43ada69be]

I (4795) BT_BTM: BTM_InqDbRead: bd addr [4bd4a6cb1f9f]

I (4795) BT_BTM: BTM_InqDbRead: bd addr [65346099b949]

I (4805) BT_BTM: BTM_InqDbRead: bd addr [42f62300294f]

I (4805) BT_BTM: BTM_InqDbRead: bd addr [410b2ac2c4b4]

I (4815) BT_BTM: BTM_InqDbRead: bd addr [68963bb24608]

I (4825) BT_BTM: BTM_InqDbRead: bd addr [6854f7e891f4]

I (4825) BT_BTM: BTM_InqDbRead: bd addr [7fa4c7232dd6]

I (4835) BT_BTM: BTM_InqDbRead: bd addr [65907e7be46c]

I (4835) BT_BTM: BTM_InqDbRead: bd addr [4645fe2509b7]

I (4845) BT_BTM: BTM_InqDbRead: bd addr [650d658f18d3]

I (4855) BT_BTM: BTM_InqDbRead: bd addr [504218c9b55d]
[... 783 lines cut ...]
rc event 2
I (19485) RC_CT: AVRC metadata rsp: attribute id 0x1, 擦肩而过
rc event 2
METADATA
I (19495) RC_CT: AVRC metadata rsp: attribute id 0x2, 宇桐非/胡雯
rc event 4
METADATA
I (19505) RC_CT: AVRC metadata rsp: attribute id 0x4, 与你同飞
METADATA
I (19505) RC_CT: AVRC metadata rsp: attribute id 0x20,
CHANGE_NOTIFY
I (19515) RC_CT: AVRC event notification: 1
I (19525) BT_AV: Playback status changed: 0x2
I (19525) BT_AVRC: AVRC_BldCommand: pdu=31 status=4
I (19535) BT_AVRC: avrc_bld_init_cmd_buffer: pdu=31, opcode=0
I (19535) BT_AVRC: avrc_bld_vol_change
I (19545) BT_AVRC: AVRC_BldCommand: returning 4
I (19545) BT_AVCT: AVCT_MsgReq
I (19555) BT_AVCT: len: 15
I (19555) BT_L2CAP: L2CA_DataWrite()  CID: 0x0043  Len: 18
I (22135) BT_BTM: BTM_SetPowerMode: pm_id 0 BDA: f38299fe mode:0x2
W (22135) BT_HCI: hci cmd send: sniff: hdl 0x80, intv(400 800)
W (22265) BT_HCI: hcif mode change: hdl 0x80, mode 2, intv 800, status 0x0
I (22265) BT_AV: ESP_BT_GAP_MODE_CHG_EVT mode: 2
```

shangke1112, 2024-03-26:
We reproduced it ourselves and found that using esp_restart() can automatically connect.
Please resend a new  log  with the default menuconfig.

**reset the board using the esp_restart() function,  then start Bluetooth to reconnect the phone,  and  do not re-pair.**

I want to know why the automatic connection failed.

jjj19950710, 2024-03-27:
I used the default configuration file before.It can be connected, just need to click pairing again on the phone.

shangke1112, 2024-03-29:
@jjj19950710  
Sorry for the misunderstanding.

**esp_bt_gap_remove_bond_device(s_peer_bda) ;** 

This function will remove the device info ,and will be cause re-paring.

**I (9945) BT_AV: Found a target device, address 20:04:f3:82:99:fe, name HONOR**

The log means that this function has been called .

closed 2025-03-06 by Alvin1Zhang