espressif/esp-idf #13417: 5.2.1 anti rollback can't be enabled (build breaks) if silent reboot panic handler enabled

greenaddress, 2024-03-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Expected behavior is for the build to be successful with any panic handling   (print registers and reboot or just silent reboot for example). 

### What is the actual behavior?

The actual behavior is that the builds fails with anti roll back enabled and silent reboot panic handling at the same time.

If anti roll back is disabled or if the panic handling is changed to print register and reboot then the build is successful. 

### Steps to reproduce.

1. copy recursively somewhere ~/esp/esp-idf/examples/get-started/hello_world/ as a base
2. create in the hello_world directory you just copied sdkconfig.defaults with the following contents:
```
CONFIG_BOOTLOADER_LOG_LEVEL_NONE=y                                                                                                                                                                                 
CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y                                         
CONFIG_BOOTLOADER_APP_ANTI_ROLLBACK=y                                           
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y                                                
CONFIG_PARTITION_TABLE_CUSTOM=y                                                 
CONFIG_PARTITION_TABLE_OFFSET=0x9000                                            
CONFIG_ESP_SYSTEM_PANIC_SILENT_REBOOT=y    
```
4. Create a partitions.csv file in the same directory with the following content:
```
# Espressif ESP32 Partition Table                                                                                                                                                                                  
# Name,   Type, SubType, Offset,  Size, Flags                                   
nvs,      data, nvs,     0xA000,  0x4000,                                       
otadata,  data, ota,     0xE000,  0x2000, encrypted                             
ota_0,    app,  ota_0,   ,         1984K,                                       
ota_1,    app,  ota_1,   ,         1984K,                                       
nvs_key,  data, nvs_keys,,            4K, encrypted  
```
5. Run idf.py build, you will get an error like the following:
```
In file included from /home/user1/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/sys-include/sys/reent.h:458,
                 from /home/user1/esp/esp-idf/components/newlib/platform_include/sys/reent.h:22,
                 from /home/user1/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/sys-include/string.h:11,
                 from /home/user1/esp/esp-idf/components/esp_system/startup.c:8:
/home/user1/esp/esp-idf/components/esp_system/startup.c: In function 'do_core_init':
/home/user1/esp/esp-idf/components/esp_system/startup.c:353:43: error: implicit declaration of function 'esp_app_get_description' [-Werror=implicit-function-declaration]
  353 |     assert(esp_efuse_check_secure_version(esp_app_get_description()->secure_version) == true && "Incorrect secure version of app");
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~       
/home/user1/esp/esp-idf/components/esp_system/startup.c:353:68: error: invalid type argument of '->' (have 'int')                                                                                                  
  353 |     assert(esp_efuse_check_secure_version(esp_app_get_description()->secure_version) == true && "Incorrect secure version of app");

```

### Build or installation Logs.

```plain
In file included from /home/user1/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/sys-include/sys/reent.h:458,
                 from /home/user1/esp/esp-idf/components/newlib/platform_include/sys/reent.h:22,
                 from /home/user1/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/sys-include/string.h:11,
                 from /home/user1/esp/esp-idf/components/esp_system/startup.c:8:
/home/user1/esp/esp-idf/components/esp_system/startup.c: In function 'do_core_init':
/home/user1/esp/esp-idf/components/esp_system/startup.c:353:43: error: implicit declaration of function 'esp_app_get_description' [-Werror=implicit-function-declaration]
  353 |     assert(esp_efuse_check_secure_version(esp_app_get_description()->secure_version) == true && "Incorrect secure version of app");
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~       
/home/user1/esp/esp-idf/components/esp_system/startup.c:353:68: error: invalid type argument of '->' (have 'int')                                                                                                  
  353 |     assert(esp_efuse_check_secure_version(esp_app_get_description()->secure_version) == true && "Incorrect secure version of app");
```

### More Information.

if you change the sdkconfig.defaults (remember to delete sdkconfig) to
```
CONFIG_BOOTLOADER_LOG_LEVEL_NONE=y                                                                                                                                                                                 
CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y                                         
CONFIG_BOOTLOADER_APP_ANTI_ROLLBACK=y                                           
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y                                                
CONFIG_PARTITION_TABLE_CUSTOM=y                                                 
CONFIG_PARTITION_TABLE_OFFSET=0x9000
```

then it works again.

Equally if app roll back is disabled then the build succeeds.

KonstantinKondrashov, 2024-03-20:
Hi @greenaddress!
Thanks for the report. I confirm that it happens on the 5.2 only, the master and <= 5.1 do not have such an issue.
If you need a fix now - [29770.patch](https://github.com/espressif/esp-idf/files/14672030/29770.patch). It should be merged soon.

mahavirj, 2024-03-26:
Fixed with https://github.com/espressif/esp-idf/commit/00f2e33008206e2ca39eb07f916931d89830b880

closed 2024-03-26 by mahavirj