espressif/esp-idf #13418: Error while building: ValueError: I/O operation on closed pipe!

masterli19, 2024-03-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.2.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Continue to build whole project after generating partion table

### What is the actual behavior?

After the partion table was generated, the build crashed.
When I typed Ctrl+C to end the build, it showed something like:

> Aborted!
> Exception ignored in: <function BaseSubprocessTransport.__del__ at 0x000002B2E5A1C4A0>
> Traceback (most recent call last):
>   File "asyncio\base_subprocess.py", line 125, in __del__
>   File "asyncio\base_subprocess.py", line 78, in __repr__
>   File "asyncio\proactor_events.py", line 80, in __repr__
>   File "asyncio\windows_utils.py", line 102, in fileno
> ValueError: I/O operation on closed pipe
> Exception ignored in: <function _ProactorBasePipeTransport.__del__ at 0x000002B2E5A1DB20>
> Traceback (most recent call last):
>   File "asyncio\proactor_events.py", line 116, in __del__
>   File "asyncio\proactor_events.py", line 80, in __repr__
>   File "asyncio\windows_utils.py", line 102, in fileno
> ValueError: I/O operation on closed pipe
> Exception ignored in: <function _ProactorBasePipeTransport.__del__ at 0x000002B2E5A1DB20>
> Traceback (most recent call last):
>   File "asyncio\proactor_events.py", line 116, in __del__
>   File "asyncio\proactor_events.py", line 80, in __repr__
>   File "asyncio\windows_utils.py", line 102, in fileno
> ValueError: I/O operation on closed pipe

I had reinstalled the esp-idf, and tried v5.1.2, 5.2.0, and 5.2.1, never succeed.
I had removed esp-adf components from my computer, it might be the course.

### Steps to reproduce.

1. Step
2. Step
3. Step
...
1. Install esp-idf v5.1.2
2. install esp-adf from github or gitee
3. remove esp-adf components
4. reinstall esp-idf v5.1.2 or v5.2.1, or other v5 branches
5. enter any project directory and run idf.py build

### Build or installation Logs.

```plain
[... 64 lines cut ...]
Exception ignored in: <function BaseSubprocessTransport.__del__ at 0x000002B2E5A1C4A0>
Traceback (most recent call last):
  File "asyncio\base_subprocess.py", line 125, in __del__
  File "asyncio\base_subprocess.py", line 78, in __repr__
  File "asyncio\proactor_events.py", line 80, in __repr__
  File "asyncio\windows_utils.py", line 102, in fileno
ValueError: I/O operation on closed pipe
Exception ignored in: <function _ProactorBasePipeTransport.__del__ at 0x000002B2E5A1DB20>
Traceback (most recent call last):
  File "asyncio\proactor_events.py", line 116, in __del__
  File "asyncio\proactor_events.py", line 80, in __repr__
  File "asyncio\windows_utils.py", line 102, in fileno
ValueError: I/O operation on closed pipe
Exception ignored in: <function _ProactorBasePipeTransport.__del__ at 0x000002B2E5A1DB20>
Traceback (most recent call last):
  File "asyncio\proactor_events.py", line 116, in __del__
  File "asyncio\proactor_events.py", line 80, in __repr__
  File "asyncio\windows_utils.py", line 102, in fileno
ValueError: I/O operation on closed pipe
PS D:\ESP32_Workspace\hello_world>
```

### More Information.

before the esp-adf installing, everything went well on esp-idf v5.1.2

dobairoland, 2024-04-02:
We will take a look at solving/improving the error message. But that won't change the fact of the build being broken.

Have you tried to `ccache -C` and `idf.py fullclean` or your project?

masterli19, 2024-04-07:
I tried idf.py fullclean after a crack, but that didn't make any difference.

closed 2024-04-17 by espressif-bot