espressif/esp-idf #13420: esp32s3 ADC converter has glitches on certain bit boundaries

jrahlf, 2024-03-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We use an esp32s3 and its inbuilt ADC for a control loop. We noticed, that the control loop achieves much better results (more precise / less noisy) around certain ADC values compared to others.

While debugging this, I noticed that the esp32s3 ADC has conversion glitches whenver the 9th, 6th, 4th or 0th bit becomes set.
Attached is a measurement of a 0.2Hz ramp from 0V to 1V. The ADC is running in the lowest range. The sampling frequency is 18.8kHz (arbitrary, same results for other frequencies). The ADC is ADC0 and the input GPIO is GPIO9. ADC calibration is not used, as this cannot help here anyway.

The graph shows there are glitches around every 16 (yellow), 64 (green), 512 (red) ADC counts (glitches for 0th bit are not shown). We tested this with multiple esp32s3 microcontrollers.

[image]
[image]

1. Can you confirm the behaviour?
2. Can you add the problem to the errata sheet?

jrahlf, 2024-03-20:
I forgot to add, there is a 100Ohm + 100nF (=16kHZ) RC lowpass close to the input pin of the module. This should not be relevant to the problem, though.

jrahlf, 2024-03-21:
So the esp32s3 datasheet specifies a DNL of 4LSB for the ADC. This means the glitches ("missing codes") are actually within spec.
However, it would still be nice if this was documented, as this seems systematic.

TD-er, 2024-03-21:
Do you have calibration enabled?
Can this be a rounding error when applying calibration?

jrahlf, 2024-03-25:
Calibration loses 2 bits of precision, the input  values 0-4096 are mapped to integers from 0-1000. It is hardly, if at all, visible after passing the data through the calibration driver.

TD-er, 2024-03-25:
I can imagine it might result in more 'jumpy' behavior when it is done doing just integer math.
But I don't know how the calibration is done internally, so just speculating.

I do know the modern ESP32-variants do not just have a single point calibration (thus linear map), but some even have a non-linear calibration curve present (or at least the functions suggest this).