espressif/esp-idf #13422: LP-Core read ADC

lbernstone, 2024-03-19:
### Is your feature request related to a problem?

I have a use case (battery charging) where it would be very helpful to be able to read the ADC on rtc pins during deep sleep by the LP-core. Is this a todo item, or is the device not capable of reading SAR ADC during deep sleep?

### Describe the solution you'd like.

ulp_lp_adc.h/.c with equivalent functionality to the riscv ulp's

ESP-Marius, 2024-03-20:
Hi,

Unfortunately C6 do not have an ADC in the LP/RTC domain that can be used during sleep. The good news it that P4 does though, so lp-core ADC support is on our roadmap.

lbernstone, 2024-03-20:
Oh, I'm a lot more likely to switch to an S2 than the P4. Cheap + wifi...

closed 2024-03-20 by lbernstone

X-Ryl669, 2024-12-07:
@ESP-Marius 
Why does the LP's GPIO are all ADC capable (and none of the other GPIO are) in that case ? Seems like it's a dumb choice, since the number of GPIO for LP core is very limited and any ADC requirement will consume the precious GPIO.

When you say it doesn't have it, do you mean that the ADC peripheral isn't powered on while (deep) sleeping ? Or that the 
LP core can't use the ADC peripheral registers ? 

Because in the former case, maybe it's possible to avoid powering down the peripheral required domains and still use it (not sure if it's a gain by se compared to HP core, probably).

ESP-Marius, 2024-12-08:
@X-Ryl669 What I mean is that there is no dedicated LP/RTC ADC. On chips that have this, e.g. S3 or P4, there is a dedicated part of the ADC HW that is meant to be used during sleep, with limited functionality, clocked by clocks available during sleep, powered by the LP/RTC power domain etc.

> maybe it's possible to avoid powering down the peripheral required domains 

It could technically be possible to tune the PMU to allow everything that is needed for the ADC to stay online during deep sleep (although i am no expert on the PMU...), but it would probably require a large part of the chip to stay powered on.

And as you say, at this point the gains are likely limited and you might aswell periodically wake up the HP CPU to perform the measurements.