espressif/esp-idf #13423: IEEE 1588 Precision Time Protocol -- can I get some feedback from Espressif?

SolidStateLEDLighting, 2024-03-20:
### Is your feature request related to a problem?

I have been asked by a client to add IEEE 1588 PTP to the Esp32 Ethernet/NetIf stack.    I'm guessing this company intends to leverage high speed services over Ethernet and the timing requirement is the first step in adding higher speed services (which might also relate to network transmission of audio, video, and other forms of time-sensitive data).

Can I get some direction from the Espressif team on how to add this work to the IDF?    I have about 5 years of experience in on the IDF and I think that I'm really coming up to speed on it.   I suspect that I might be of some value to you if I gain just a bit of insight on how to integrate the generation/reception of custom Ethernet packets within the framework.

For your reference -- here is my public offering to the community:
https://github.com/SolidStateLEDLighting/esp32-S3-idf-advanced-template-alpha-5.2
My direct contact information is there as well.

Thanks for the assistance...

kostaond, 2024-03-20:
@SolidStateLEDLighting IEEE1588 is **not supported** by ESP32. The ESP32 TRM used to list it as supported feature in past but it was a mistake. However, ESP32P4 supports IEEE1588. I'm planning to start implementing infrastructure to retrieve the time stamp to the higher layers in Ethernet driver next month. Implementing some PTP deamon is different story though... I was thinking about PTPd but it seems it is dead project. I still haven't made any final decision yet on this topic. If you had any recommendation, it's very welcomed.

SolidStateLEDLighting, 2024-03-20:
If I volunteered to give you a hand with some of this, would Espressif be willing to put some P4 hardware in my hands?  (I'm in the Philippines and I would pay for shipping to bring it here).

I'm not new to firmware development, but I'm just coming up to speed with PTP and the related protocols, so I don't currently have a strong recommendation except to say that AVB (Audio Visual Bridging - IEEE 802.1AS) might be a better first level goal because PTP was incorporated into that standard.  (PTP might be considered a more generic standard).     The AVB also leads to Audio and Video streaming which might be significantly more interesting to Espessif.

The client who is considering hiring me to pursue this effort, might be an excellent launch customer for the P4 and he is willing to put some money into it if he thinks I can reach his targets.    I'm not permitted to say much more about his project because he is in a competitive market and this investment is a gamble for him.

Is there any particular reason that the Esp32 won't perform well enough to handle PTP?   Or is this just a management decision at Espressif not to backport this into the Esp32?     It's a shame that the newer hardware doesn't contain the MAC for Ethernet support.

kostaond, 2024-03-20:
I was also thinking about implementing deamon based on gPTP (IEEE Std 802.1AS).

ESP32 does not support hardware time stamping in MAC module. If you done it at SW layer, you would never achieved required precision.

Regarding your offer, I would need to discuss internally...

SolidStateLEDLighting, 2024-03-20:
I think you may be on the right track with gPTP (IEEE Std 802.1AS).

My client is likely to be very motivated to help.   He doesn't not know the price point on the P4, so he wouldn't know how to evaluate it on a cost basis, but he wants the functionality that can deliver an audio or video capable streaming network. (and of course this requires the precision time service)

Another issue will be how long before the P4 will be deemed ready for market.   He would want to get his product to market in a time frame that is not too far off in the future.   Have your people completed the validation of the silicon?

Please let me know what we can do on the P4 hardware and I'll continue to assess how I can help you on this end.   Thanks.

kostaond, 2024-03-20:
I'm of course not authorized to provide any sales related details. Please contact as [here](https://www.espressif.com/en/contact-us/sales-questions).

Just one important note about ESP32P4, it does not come with WiFi module in SoC. However, there is a solution planned to cover it. There will be option to have ESP32P4 with bound with other ESP WiFi capable chip.

SolidStateLEDLighting, 2024-03-20:
I'm not too worried about pricing myself.   He just can't judge it until those numbers come out.  I guess he will take a guess at it.

Yes, I am aware that the P4 does not support wireless right now.   Ethernet is what is required for the high-speed processing that he is interested in.   Thanks.

SolidStateLEDLighting, 2024-03-20:
For an Ethernet wired network -- I believe the PHY is the only place fast enough to hold an accurate time-stamp.    I was reviewing some documentation on this and at least one PHY that support this feature uses a Start of Frame detection to improve its compliance with the 1588v2 PTP specification.    Once the PHY has that time value, then I assume it raises in interrupt for the processor to work with that value through RMII.

So, this explains why my client is open to the idea of using the Esp32 for a precision timing application.

Did Espressif integrate a PHY into the P4 directly?   OR -- does your testing hardware PHY not have this time-stamp capability?

Can you clarify how the P4 would be so close to the network that it could record a time-stamp without software?   Thanks.

SolidStateLEDLighting, 2024-03-22:
Today is Friday.   Is there any word on if I might be able to get a couple of samples of the P4?    I assume this would be just a raw chip.

kostaond, 2024-03-22:
I'm not in position to obtain you samples. Please visit us [here](https://www.espressif.com/en/contact-us/get-samples).

TD-er, 2024-03-22:
> For an Ethernet wired network -- I believe the PHY is the only place fast enough to hold an accurate time-stamp. I was reviewing some documentation on this and at least one PHY that support this [...]

I think the RMII Ethernet devices are probably your best bet here (assuming the P4 will have RMII available) as it allows for quite low-level access to the interface.
Some SPI Ethernet devices do a lot themselves (e.g. W5500), meaning such timing capabilities should already be present in those chips or else you will have to accept quite a high jitter compared to RMII Ethernet chips.

Other ideas how it might be possible to implement a precision time sync via WiFi is by looking into the _Fine Time Measurement_ (FTM) protocol, which is supported by a number of ESP32-variants (e.g. ESP32-C6)
You can also get the TSF time from the AP you're connected to.
In my network my APs report their uptime when I query their TSF time.
Maybe you can use this as some sync among nodes connected to the same AP?

SolidStateLEDLighting, 2024-03-22:
Ok, guys -- thanks for the advice.

kostaond, 2024-11-08:
@SolidStateLEDLighting, it took a while but today, the very first version of time synchronization was merged in our internal GitLab repo! Stay tuned! It will be soon synchronized with the GitHub repo.

closed 2024-11-08 by kostaond

SolidStateLEDLighting, 2024-11-08:
Great job!   That is going to be a valuable feature.

skylin008, 2025-01-23:
Is there any source code to be view about ieee1588 v2, thanks!

reinierweerts, 2025-05-08:
I have implemented the PTPD and in components/soc/esp32p4/include/soc/emac_ptp_struct.h see some references to PPS0-3.
In the P4 TRM I can't find a signal that would route a PPS pulse to either a GPIO pin (for oscilloscope) or to MCPWM capture. Is there / will there be any such function?

kostaond, 2025-05-09:
@reinierweerts there is no such function available in current EMAC version. However, we've already asked our digital team to implement(enable) it in future chips.

I toggle GPIO directly from ISR in our [ptp example](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/ptp). It's not ideal but at least something...

BTW: you have implemented [ptpd](https://github.com/ptpd/ptpd)?