espressif/esp-idf #13427: Nimble BLE in SDK v5.2.1 TX power very low and unstable.

filzek, 2024-03-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The Nimble BLE i. Blufi or in SPP always have different tx power upon boot. Sometimes extreme high power, sometimes very low power, no matter I'd we define the tx power it simple doesn't respect it.

So BLE TX power is broken in sdk v5.2.1

filzek, 2024-03-23:
found something quite interesting, it works if you set the power twice, one right after nimble init and another after all things are set, so there is a but in between the initialization and in the after the system ready.

xiongweichao, 2024-03-25:
Hi @filzek ， May I ask which chip you use?

filzek, 2024-03-27:
ESP32 Rev 3.0,

xiongweichao, 2024-07-10:
Hi @filzek ,
I'm very sorry to reply you so late. Would it work if you set it only after nimble init? Thanks!

closed 2025-01-17 by filzek