espressif/esp-idf #13428: examples/network/sta2eth doesn't connect

owenthewizard, 2024-03-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1-192-g7e36e97467

### Espressif SoC revision.

ESP32-D0WD (revision v1.0)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

WT32-ETH01 V1.2

### Power Supply used.

USB

### What is the expected behavior?

I expect the SSID and password submitted via HTTP (http://wifi.settings) to be used to connect to the access point.
```
I (10301) NCM_configuration: Found URL query => ssid=My+Real+Wifi&password=my+spaced+password
I (10311) NCM_configuration: ssid=My+Real+Wifi
I (10311) NCM_configuration: password=my+spaced+password
I (10321) NCM_configuration: WiFi settings accepted!
```

### What is the actual behavior?

```
I (5375) example_sta2wired: Wi-Fi STA disconnected
I (7785) example_sta2wired: Wi-Fi STA disconnected
I (10205) example_sta2wired: Wi-Fi STA disconnected
E (10555) example_sta2wired: WiFi station connected failed
```

### Steps to reproduce.

1. Build and flash examples/network/sta2eth
2. Connect to Ethernet on host (e.g. laptop), navigate to http://wifi.settings
3. Input the SSID and password fo your AP.
4. ESP32 reboots as expected but does not connect.

### Debug Logs.

```plain
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5940
load:0x40078000,len:15100
load:0x40080400,len:4
[... 289 lines cut ...]
I (395) wifi:Init dynamic tx buffer num: 32
I (405) wifi:Init static rx buffer size: 1600
I (405) wifi:Init static rx buffer num: 10
I (405) wifi:Init dynamic rx buffer num: 32
I (415) wifi_init: rx ba win: 6
I (415) wifi_init: tcpip mbox: 32
I (425) wifi_init: udp mbox: 6
I (425) wifi_init: tcp mbox: 6
I (425) wifi_init: tcp tx win: 5760
I (435) wifi_init: tcp rx win: 5760
I (435) wifi_init: tcp mss: 1440
I (445) wifi_init: WiFi IRAM OP enabled
I (445) wifi_init: WiFi RX IRAM OP enabled
I (455) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (455) example_sta2wired: Starting USB-WiFi bridge
I (465) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
I (545) wifi:mode : sta (7c:87:ce:44:a9:a8)
I (545) wifi:enable tsf
I (2965) example_sta2wired: Wi-Fi STA disconnected
I (5375) example_sta2wired: Wi-Fi STA disconnected
I (7795) example_sta2wired: Wi-Fi STA disconnected
I (10205) example_sta2wired: Wi-Fi STA disconnected
E (10555) example_sta2wired: WiFi station connected failed
(reboot and start again)
```

kostaond, 2024-05-02:
[owenthewizard](https://github.com/owenthewizard) I tried to reproduce and the example works as expected. Did you fill correct SSID and password?

owenthewizard, 2024-05-02:
Did you try with spaces or other characters in the SSID/password? I still have the issue, tested with my main SSID (contains spaces) and password (contains spaces), as well as my iPhone hotspot (in both maximize compatibility and regular mode), SSID has special character `’` as well as a space, password has no spaces.

kostaond, 2024-05-03:
I see,that's important piece of information! Thanks for pointing it out. It really doesn't work when space is present either in SSID or password. The spaces are replaced by pluses (+) in URL query but they are not decoded back to spaces. We'll have a look at it.

closed 2026-09-24 by espressif-bot