espressif/esp-idf #13430: esp_http_server: Adding custom HTTP headers in async requests causes crash

tmiw, 2024-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I expected to be able to set custom HTTP headers inside a request object created by `httpd_req_async_handler_begin` and send the response to the browser without any crashes.

### What is the actual behavior?

Board reboots while inside the task handling the request (not the HTTP server task).

### Steps to reproduce.

Add code similar to the following inside registered `esp_http_server` handler:

```c++
ESP_ERROR_CHECK(httpd_resp_set_type(req, "text/html"));
ESP_ERROR_CHECK(httpd_resp_set_hdr(req, "Content-Security-Policy", "script-src 'self' 'unsafe-inline'"));

[...]

httpd_req_t* asyncReq;
esp_err_t err = httpd_req_async_handler_begin(req, &asyncReq);
if (err == ESP_OK)
{
    // queue request on other thread
}
```

and then inside the function in the async task handling the request:

```c++
if (httpd_resp_send_chunk(message->request, chunk, chunksize) != ESP_OK) 
{
    ...
}
```

Note that moving `httpd_resp_set_*` to the async task had no effect on this bug.
[... 75 lines cut ...]
+    // Copy response header block
+    struct httpd_data *hd = (struct httpd_data *) r->handle;
+    struct httpd_req_aux* asyncAux = (struct httpd_req_aux*)async->aux;
+    struct httpd_req_aux* stdAux = (struct httpd_req_aux*)r->aux;
+
+    asyncAux->resp_hdrs = calloc(hd->config.max_resp_headers, sizeof(struct resp_hdr));
+    if (asyncAux->resp_hdrs == NULL)
+    {
+        free(asyncAux);
+        free(async);
+        return ESP_ERR_NO_MEM;
+    }
+    memcpy(asyncAux->resp_hdrs, stdAux->resp_hdrs, hd->config.max_resp_headers * sizeof(struct resp_hdr));
+
     // mark socket as "in use"
     struct httpd_req_aux *ra = r->aux;
     ra->sd->for_async_req = true;
@@ -593,6 +607,7 @@ esp_err_t httpd_req_async_handler_complete(httpd_req_t *r)
     struct httpd_req_aux *ra = r->aux;
     ra->sd->for_async_req = false;
 
+    free(ra->resp_hdrs);
     free(r->aux);
     free(r);
```

tmiw, 2024-03-21:
Created this bug for traceability and since I'm not 100% sure if the change I tried above is the correct one. If it looks reasonable, I can go ahead and create a PR.

chipweinberger, 2024-03-21:
Youre changes look reasonable to me.

Btw, I was the one that implemented async requests in ESP-IDF! =) glad people are using them!

tmiw, 2024-03-25:
> Your changes look reasonable to me. You should open a PR!
> 
> Btw, I was the one that implemented async requests in ESP-IDF! =) glad people are using them!

Nice work! I just created #13462 with the above change :)

mahavirj, 2024-03-26:
Closing in favour of https://github.com/espressif/esp-idf/pull/13462

closed 2024-03-26 by mahavirj

hmalpani, 2024-03-29:
Hello @tmiw 
I was trying to create the issue. I have done these changes in async_handler example:
```
static esp_err_t submit_async_req(httpd_req_t *req, httpd_req_handler_t handler)
 {
+    ESP_ERROR_CHECK(httpd_resp_set_type(req, "text/html"));
+    ESP_ERROR_CHECK(httpd_resp_set_hdr(req, "Content-Security-Policy", "script-src 'self' 'unsafe-inline'"));
     // must create a copy of the request that we own
     httpd_req_t* copy = NULL;
```
```
@@ -133,8 +135,8 @@ static esp_err_t long_async_handler(httpd_req_t *req)
     // send a request count
     char s[100];
     snprintf(s, sizeof(s), "<div>req: %u</div>\n", req_count);
-    httpd_resp_sendstr_chunk(req, s);
-
+    // httpd_resp_sendstr_chunk(req, s);
+    httpd_resp_send_chunk(req, s, strlen(s));
     // then every second, send a "tick"
     for (int i = 0; i < 60; i++) {
```

I am not observing the crash in this case. Can you please let me know if these are the changes that created the issue on your side.

tmiw, 2024-03-30:
The example seemed to work for me as well once I increased the stack size of the async tasks.

Anyway, you can try the code for my project (https://github.com/tmiw/ezDV/tree/1.0.2/firmware) with the following patch to disable everything but the HTTP-related logic (and reenable async handling of HTTP requests):

```diff
diff --git a/firmware/main/Application.cpp b/firmware/main/Application.cpp
index 17b6378..34b103a 100644
--- a/firmware/main/Application.cpp
+++ b/firmware/main/Application.cpp
@@ -219,6 +219,7 @@ void App::onTaskStart_()
 {
     ESP_LOGI(CURRENT_LOG_TAG, "onTaskStart_");
 
+#if 0
     // Enable peripheral power.
     enablePeripheralPower_();
     
@@ -372,19 +373,20 @@ void App::onTaskStart_()
             uiTask_ = new ui::UserInterfaceTask();
             assert(uiTask_ != nullptr);
             start(uiTask_, pdMS_TO_TICKS(1000));
-        
+#endif
+
             // Start Wi-Fi
             wirelessTask_ = new network::WirelessTask(freedvTask_, tlv320Device_, audioMixer_, voiceKeyerTask_);
             assert(wirelessTask_ != nullptr);
             
-            wirelessTask_->setWiFiOverride(wifiOverrideEnabled_);
+            wirelessTask_->setWiFiOverride(true);
             start(wirelessTask_, pdMS_TO_TICKS(5000));
 
             // Start storage handling
             settingsTask_ = new storage::SettingsTask();
             assert(settingsTask_ != nullptr);
             settingsTask_->start();
-            
+#if 0            
             softwareUpdateTask_ = new storage::SoftwareUpdateTask();
             assert(softwareUpdateTask_ != nullptr);
             softwareUpdateTask_->start();
@@ -411,6 +413,7 @@ void App::onTaskStart_()
             start(rfComplianceTask_, pdMS_TO_TICKS(1000));
         }
     }
+#endif
 }
 
 void App::onTaskSleep_()
diff --git a/firmware/main/network/HttpServerTask.cpp b/firmware/main/network/HttpServerTask.cpp
index 611b5fa..98051e8 100644
--- a/firmware/main/network/HttpServerTask.cpp
+++ b/firmware/main/network/HttpServerTask.cpp
@@ -319,6 +319,7 @@ esp_err_t HttpServerTask::ServeStaticPage_(httpd_req_t *req)
     ESP_LOGI(CURRENT_LOG_TAG, "Sending file : %s (%ld bytes)...", filename, file_stat.st_size);
     set_content_type_from_file(req, filename);
 
+#if 0
     char* scratchBuf = (char*)heap_caps_malloc(SCRATCH_BUFSIZE, MALLOC_CAP_SPIRAM | MALLOC_CAP_32BIT);
     assert(scratchBuf != nullptr);
 
@@ -344,8 +345,8 @@ esp_err_t HttpServerTask::ServeStaticPage_(httpd_req_t *req)
 
     ESP_LOGI(CURRENT_LOG_TAG, "Sending %s complete", filename);
     return ESP_OK;
+#endif
 
-#if 0
     // Async requests are disabled due to ESP-IDF bug with additional HTTP headers.
     // See https://github.com/espressif/esp-idf/issues/13430.
[... 53 lines cut ...]

Backtrace: 0x400556d2:0x3fcd6bf0 0x420623cd:0x3fcd6c00 0x42020b5f:0x3fcd6c50 0x420f5241:0x3fcd6cb0 0x420f5189:0x3fcd6cd0 0x42016ea9:0x3fcd6cf0 0x42016ef5:0x3fcd6d40 0x42016f71:0x3fcd6d60 0x40384631:0x3fcd6d80
0x400556d2: strlen in ROM
0x420623cd: httpd_resp_send_chunk at /Users/mooneer/esp-idf/components/esp_http_server/src/httpd_txrx.c:338
0x42020b5f: ezdv::network::HttpServerTask::onHttpServeStaticFileMessage_(ezdv::task::DVTask*, ezdv::network::HttpServerTask::HttpServeStaticFileMessage*) at /Users/mooneer/ezDV/firmware/main/network/HttpServerTask.cpp:244
0x420f5241: ezdv::task::DVTask::MessageFnPtrStorage<ezdv::network::HttpServerTask, ezdv::network::HttpServerTask::HttpServeStaticFileMessage>::call(ezdv::task::DVTask*, ezdv::network::HttpServerTask::HttpServeStaticFileMessage*) at /Users/mooneer/ezDV/firmware/main/task/DVTask.h:205 (discriminator 4)
0x420f5189: void ezdv::task::DVTask::HandleEvent_<ezdv::network::HttpServerTask::HttpServeStaticFileMessage>(void*, char const*, long, void*) at /Users/mooneer/ezDV/firmware/main/task/DVTask.h:335
0x42016ea9: ezdv::task::DVTask::singleMessagingLoop_(long long) at /Users/mooneer/ezDV/firmware/main/task/DVTask.cpp:341 (discriminator 2)
0x42016ef5: ezdv::task::DVTask::threadEntry_() at /Users/mooneer/ezDV/firmware/main/task/DVTask.cpp:361
0x42016f71: ezdv::task::DVTask::ThreadEntry_(ezdv::task::DVTask*) at /Users/mooneer/ezDV/firmware/main/task/DVTask.cpp:392
0x40384631: vPortTaskWrapper at /Users/mooneer/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: aa3ad28aed5f751f

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42101486
0x42101486: offset_to_block at /Users/mooneer/esp-idf/components/heap/tlsf/tlsf_block_functions.h:79
 (inlined by) block_next at /Users/mooneer/esp-idf/components/heap/tlsf/tlsf_block_functions.h:92
 (inlined by) tlsf_walk_pool at /Users/mooneer/esp-idf/components/heap/tlsf/tlsf.c:772
```

Hopefully this helps with your testing efforts!

hmalpani, 2024-04-16:
Hello @tmiw 
I am unable to build the example you shared. Can you create a simple example using ESP-IDF which can help recreate the issue? As of now, I am unable to recreate the issue.

tmiw, 2024-04-17:
> Hello @tmiw I am unable to build the example you shared. Can you create a simple example using ESP-IDF which can help recreate the issue? As of now, I am unable to recreate the issue.

I suspect the submodules and third party libraries my project normally uses are part of why you were having problems. I went ahead and created a new branch in my project that does the following:

1. Removes all non-firmware code entirely.
2. Removes all Git submodules and references to third-party libraries in the CMake files.
3. Removes all code except for the bare minimum needed to duplicate the issue.
4. Disables PSRAM and removes unneeded partitions (allowing usage on pretty much every ESP32-S3 dev board).

To run:

```
$ git clone -b esp-idf-13430 https://github.com/tmiw/ezDV.git esp-httpd-crash-demo
$ cd esp-httpd-crash-demo
$ idf.py build flash monitor

# Connect to the "ezDV xxxx" network, then go to http://192.168.4.1/ in your browser
```

which results in the following crash for me:

```
I (16262) HttpServerTask: Sending file : /index.html (25429 bytes)...
I (16632) HttpServerTask: Sending file : /bootstrap.min.css.gz (27324 bytes)...
I (16662) HttpServerTask: Sending file : /localstyle.css (3249 bytes)...
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400556d5  PS      : 0x00060c30  A0      : 0x8202c1d8  A1      : 0x3fcc1230  
0x400556d5: strlen in ROM

A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3fcc1200  
A10     : 0x00000002  A11     : 0x00000034  A12     : 0x3c0ad428  A13     : 0x00000002  
A14     : 0x00000000  A15     : 0x00000002  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  
0x400556d5: strlen in ROM
0x400556e5: strlen in ROM

Backtrace: 0x400556d2:0x3fcc1230 0x4202c1d5:0x3fcc1240 0x4200ab4b:0x3fcc1290 0x4208be41:0x3fcc12f0 0x4208be19:0x3fcc1310 0x420097f1:0x3fcc1330 0x42009839:0x3fcc1380 0x420098b5:0x3fcc13a0 0x4037fc85:0x3fcc13c0
0x400556d2: strlen in ROM
0x4202c1d5: httpd_resp_send_chunk at /Users/mooneer/esp-idf/components/esp_http_server/src/httpd_txrx.c:338
0x4200ab4b: ezdv::network::HttpServerTask::onHttpServeStaticFileMessage_(ezdv::task::DVTask*, ezdv::network::HttpServerTask::HttpServeStaticFileMessage*) at /Users/mooneer/ezDV/main/network/HttpServerTask.cpp:185
0x4208be41: ezdv::task::DVTask::MessageFnPtrStorage<ezdv::network::HttpServerTask, ezdv::network::HttpServerTask::HttpServeStaticFileMessage>::call(ezdv::task::DVTask*, ezdv::network::HttpServerTask::HttpServeStaticFileMessage*) at /Users/mooneer/ezDV/main/task/DVTask.h:208 (discriminator 4)
0x4208be19: void ezdv::task::DVTask::HandleEvent_<ezdv::network::HttpServerTask::HttpServeStaticFileMessage>(void*, char const*, long, void*) at /Users/mooneer/ezDV/main/task/DVTask.h:338
0x420097f1: ezdv::task::DVTask::singleMessagingLoop_(long long) at /Users/mooneer/ezDV/main/task/DVTask.cpp:346 (discriminator 2)
0x42009839: ezdv::task::DVTask::threadEntry_() at /Users/mooneer/ezDV/main/task/DVTask.cpp:366
0x420098b5: ezdv::task::DVTask::ThreadEntry_(ezdv::task::DVTask*) at /Users/mooneer/ezDV/main/task/DVTask.cpp:397
0x4037fc85: vPortTaskWrapper at /Users/mooneer/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: 08eece86af3ea9fc

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759fc
0x403759fc: esp_restart_noos at /Users/mooneer/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:159
```