espressif/esp-idf #13432: How to implement encryption and decryption of aes algorithm in secure boot

ouer1224, 2024-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi teams,
I am trying to add some new feature in the secure bootloader. And I need to use AES algorithm to decrypt and encrypt some data.
Due to the limitations of the bootloader ram, I was unable to call mbedtls and the AES functions I wrote myself.  So I want to call the esp32 hardware aes calculation function.  But I can't call 'esp_aes_crypt_ecb()' and 'esp_aes.h' in bootloader.  

Is there any way I can use AES algorithm for encryption and decryption in secure bootloader.

boarchuz, 2024-03-21:
aes_128_cbc_decrypt and aes_128_cbc_encrypt are available in ROM:
https://github.com/espressif/esp-idf/blob/4f3cd0deb9c79c8282da4938a29d265705a57564/components/esp_rom/esp32/ld/esp32.rom.redefined.ld#L8-L9

closed 2024-03-26 by mahavirj

ouer1224, 2024-03-27:
Sorry for the later reply.  I am fixing a memory release bug.  As soon as it is fixed, I will test the ROM APIs and aes_hal.c or tinycrypt.