espressif/esp-idf #13433: idf.py coredump-info sould accept .elf file as command line parameter to allow standalone usage

SukuWc, 2024-03-21:
### Is your feature request related to a problem?

I want to decode a large batch of coredump binaries retrieved from users using many versions of the firmware. For that I have the .elf file for all of the builds matched to each cordump binary.  I want to call idf.py coredump-info -c coredump.bin but it only works if I manually copy the .elf to the build folder beforehand.

### Describe the solution you'd like.

idf.py coredump-info should be able to get .elf path via command line parameter

igrr, 2024-03-21:
I think for this type of usage, you can use the [`esp-coredump`](https://github.com/espressif/esp-coredump) tool directly. `idf.py` integration is only meant to support use cases occurring in a development environment. Besides, if you use esp-coredump, you don't need to install the whole ESP-IDF on the server where you will process the coredumps.

dobairoland, 2024-03-21:
I see this advise is missing from the documentation. We'll update it....

closed 2024-04-23 by espressif-bot