espressif/esp-idf #13435: Multi-Target Build ?

diplfranzhoepfinger, 2024-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

is there a way to build same Project for multi Target with a Script ? 

like for a ESP32S3 N8R2 and N8R8

or like for a C6 and S3 ? 

so that we have 2 Binaries automatically after script run ? 

thanks,

igrr, 2024-03-21:
You can build different configs in different directories, please check this example: https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/multi_config

You still need to invoke idf.py (or CMake directly) twice, though. You can consider using the [idf-build-apps](https://github.com/espressif/idf-build-apps) tool to automate the process.

diplfranzhoepfinger, 2024-03-21:
@igrr VERY god Support, thanks.

diplfranzhoepfinger, 2024-03-21:
idf-build-apps tool i do not see the Advantage ? 

a simple Shell script would do the same right ?

diplfranzhoepfinger, 2024-03-26:
it does not work. 
[sdkconfig.defaults.N8R2.txt](https://github.com/espressif/esp-idf/files/14758083/sdkconfig.defaults.N8R2.txt)
[sdkconfig.defaults.N8R8.txt](https://github.com/espressif/esp-idf/files/14758084/sdkconfig.defaults.N8R8.txt)

i do following: 

```
idf.py -B build_N8R2 -D SDKCONFIG_DEFAULTS="sdkconfig.defaults.N8R2" build
idf.py -B build_N8R8 -D SDKCONFIG_DEFAULTS="sdkconfig.defaults.N8R8" build
```

but it builds 2 times the N8R2 Version. 
not 1 time the N8R8 and 1 time the N8R2

igrr, 2024-03-26:
Have you also specified the sdkconfig file? You need to either do the same thing as the example does:
https://github.com/espressif/esp-idf/blob/be06a6f5ffe36f9554cfc91fe2036e0fc85fea60/examples/build_system/cmake/multi_config/CMakeLists.txt#L3-L6

or pass `-DSDKCONFIG=build_N8R2/sdkconfig` (for the first build) and `-DSDKCONFIG=build_N8R8/sdkconfig` (for the 2nd) on the command line.

diplfranzhoepfinger, 2024-03-26:
aaaaaaaaaaaaaah

yes. this is the Problem.

Indastri, 2024-04-23:
@diplfranzhoepfinger were you able to resolve the issue? If so please feel free to close this issue.

Cheers!

diplfranzhoepfinger, 2024-04-23:
YES ! works good

diplfranzhoepfinger, 2024-05-23:
i have another Topic:

diplfranzhoepfinger, 2024-05-23:
If i do this: 

```
idf.py set-target esp32s3
idf.py build
idf.py set-target esp32
idf.py -B build_ESP32 build
idf.py set-target esp32s3
idf.py -B build_ESP32S3 build
```

diplfranzhoepfinger, 2024-05-24:
i understood it:

diplfranzhoepfinger, 2024-06-08:
Question: 

what about the dependencies.lock File ? 

can it also be placed into the Folders ?

diplfranzhoepfinger, 2024-06-08:
and: what about Eclipse ? 
it seems the Eclipse SDK Config editor is no more working in this Scenario.

igrr, 2024-06-08:
You can specify the location of dependencies.lock: https://github.com/espressif/esp-idf/commit/17224f325450a001c4744064470cb6813b1c8514.

Regarding the sdkconfig editor not working, could you please open an issue in https://github.com/espressif/idf-eclipse-plugin/issues and provide more details?

Alvin1Zhang, 2024-07-23:
Thanks for reporting, feel free to reopen or create a new ticket.

closed 2024-07-23 by Alvin1Zhang

diplfranzhoepfinger, 2024-07-23:
> Thanks for reporting, feel free to reopen or create a new ticket.

@Alvin1Zhang 
@igrr 

all very stable now. 
just the sdkconfig Editor in Ecplise Problem https://github.com/espressif/idf-eclipse-plugin/issues/981

all in this Ticket solved very well.