espressif/esp-idf #13436: How to install node and npm on espressif/esp-idf docker when running using CI/CD

krupis, 2024-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello. In our project we rely on some node modules. They are required in order for our project to be built sucesfully.

On our local development machines, we all have node and npm installed and we have all required node modules in package.json

So when new develper wants to work with our project, he does ```npm install``` to automatically install all required modules.

I am now trying to run build using CI/CD pipeline but I cannot figure out how can I install node and npm on the docker container that runs the CI/CD pipeline. See my pipeline job below:

```
esp_idf_generate_build_files:
  image: espressif/idf:v5.0.4
  variables:
    GIT_SUBMODULE_STRATEGY: normal
  stage: build
  rules:
    - if: $CI_COMMIT_TAG # Run this job when a tag is created
  script:
    - apt-get install nodejs
    - apt-get install npm
    - node --version
    - npm --version
    - idf.py build
```

The logs from the job:
```
Running with gitlab-runner 16.6.1 (f5da3c5a)
  on Docker runner for SystemAir IAM 4JyMxyHH, system ID: s_1de69ef5[2](https://ems-gitlab.teltonika.lt/systemair/Bifrost-IAM-V2/-/jobs/935#L2)8ed
Preparing the "docker" executor
00:0[3](https://ems-gitlab.teltonika.lt/systemair/Bifrost-IAM-V2/-/jobs/935#L3)
Using Docker executor with image espressif/idf:v5.0.4 ...
Pulling docker image espressif/idf:v5.0.4 ...
Using docker image sha256:354276f81cc8b4555c5a49ce7e35ce0fa9f546fc711012c379350b61c7c3e93e for espressif/idf:v5.0.4 with digest espressif/idf@sha256:769698ea563bb0b218654e0a430c2e21e608a650dfbbebd32efcb1d7d8cd702b ...
Preparing environment
00:00
Running on runner-4jymxyhh-project-77-concurrent-0 via ems-gitlab...
Getting source from Git repository
00:02
Fetching changes with git depth set to 50...
Reinitialized existing Git repository in /builds/systemair/Bifrost-IAM-V2/.git/
Checking out 1424a575 as detached HEAD (ref is 0.4.6)...
Removing JOB_ID.txt
Removing release_notes.md
Removing version.txt
Updating/initializing submodules with git depth set to 50...
Updated submodules
Downloading artifacts
00:01
Downloading artifacts for prepare_job (934)...
Downloading artifacts from coordinator... ok        host=ems-gitlab.teltonika.lt id=934 responseStatus=200 OK token=glcbt-64
Executing "step_script" stage of the job script
00:03
Using docker image sha256:35[4](https://ems-gitlab.teltonika.lt/systemair/Bifrost-IAM-V2/-/jobs/935#L4)276f81cc8b4[5](https://ems-gitlab.teltonika.lt/systemair/Bifrost-IAM-V2/-/jobs/935#L5)55c5a49ce7e35ce0fa9f54[6](https://ems-gitlab.teltonika.lt/systemair/Bifrost-IAM-V2/-/jobs/935#L6)fc[7](https://ems-gitlab.teltonika.lt/systemair/Bifrost-IAM-V2/-/jobs/935#L7)11012c379350b61c7c3e93e for espressif/idf:v5.0.4 with digest espressif/idf@sha256:76969[8](https://ems-gitlab.teltonika.lt/systemair/Bifrost-IAM-V2/-/jobs/935#L8)ea563bb0b218654e0a430c2e21e608a650dfbbebd32efcb1d7d8cd702b ...
Detecting the Python interpreter
Checking "python3" ...
Python 3.8.10
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Python requirements are satisfied.
Requirement files:
 - /opt/esp/idf/tools/requirements/requirements.core.txt
Python being checked: /opt/esp/python_env/idf5.0_py3.8_env/bin/python
Added the following directories to PATH:
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
  /opt/esp/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin
  /opt/esp/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin
  /opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin
  /opt/esp/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin
  /opt/esp/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin
  /opt/esp/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /opt/esp/tools/cmake/3.24.0/bin
  /opt/esp/tools/openocd-esp32/v0.12.0-esp32-2023041[9](https://ems-gitlab.teltonika.lt/systemair/Bifrost-IAM-V2/-/jobs/935#L9)/openocd-esp[32](https://ems-gitlab.teltonika.lt/systemair/Bifrost-IAM-V2/-/jobs/935#L32)/bin
  /opt/esp/python_env/idf5.0_py3.8_env/bin
  /opt/esp/idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:
  idf.py build
$ apt-get install nodejs
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package nodejs

```

What could be the issue? How can I install node and npm ?

dobairoland, 2024-03-21:
https://packages.ubuntu.com/jammy/nodejs - you can see it in the title `[security] [universe]`.

So I would try:
```sh
add-apt-repository universe
apt-get update
```

The same goes for other packages. Repository must be added if `apt` cannot find the package.

krupis, 2024-03-21:
I have found a workaround that seems to work:
```
esp_idf_generate_build_files:
  image: espressif/idf:v5.0.4
  variables:
    GIT_SUBMODULE_STRATEGY: normal
  stage: build
  rules:
    - if: $CI_COMMIT_TAG # Run this job when a tag is created
  script:
    - apt-get install curl
    - curl -sL https://deb.nodesource.com/setup_18.x  | bash -
    - apt-get install nodejs
    - node --version
    - npm --version
    - npm install
    - idf.py build
 ```

dobairoland, 2024-03-25:
Ok, great. I think we can close this.

closed 2024-03-25 by dobairoland