espressif/esp-idf #13437: Incorrect `ble_addr_t` type of scanned advertisements

robin96c, 2024-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1 (originally perceived on v5.1.1)

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

My application connects 2 esp32s3's together over BLE. One acts as a central, the other as a peripheral. I use the NimBLE host.
After a bond has been established, the central will only connect with an advertiser that is in the stored bonds list. This is done by comparing the advertiser address with the addresses in the bond list.

Central and peripheral device should always reconnect automatically after disconnecting/rebooting.

### What is the actual behavior?

I noticed the address check sometimes fails because the `ble_addr_t` MAC matches, but the `type` does not.

After initial pairing, the scanned advertisement packets show the correct type even after connecting/disconnecting multiple times.
`Address type = 0 (BLE_ADDR_PUBLIC)`
If the central reboots, the scanned advertisement packets show the correct type until a disconnect occurs. From that moment on, a different type is shown instead and my application refuses to connect to the advertiser. 
`Address type = 2 (BLE_ADDR_PUBLIC_ID)`
A reboot of the central fixes the issue (until another disconnect happens).

I did a test where I put esp32s3 central devices in scanning mode, 1 with a stored bond to the peripheral, 1 without. Both receive the same advertisement packet but report a different address type.
So it seems that address type gets filled in incorrectly for old bonds.

I cannot reproduce the issue on the esp32, so it seems related to esp32s3 only.
Weirdly enough, if I replace any of the two devices (central or peripheral) with an esp32, I cannot reproduce the issue either. Seems to only be an issue between two esp32s3 devices.

### Steps to reproduce.

1. Pair 2 esp32s3 devices
2. Reboot the central device
3. Reconnect the devices
4. Disconnect the devices
5. If you log the scanned packets, the central will now start showing a different address type of the scanned peripheral advertisement packets

rahult-github, 2024-08-07:
Hi @robin96c ,  can you please confirm if random address is being used ?. Can you please share the sdkconfig you using . 

I tried with existing bleprph / blecent example to reproduce the issue, but didn't observe the issue. 

So , can you also please some sample application that can assist in reproducing this issue ?

robin96c, 2024-08-19:
Hi @rahult-github 
 
Thanks for looking into this! I'm advertising with a public address. To better reproduce the issue, I've modified the bleprph / blecent example code. The two most notable changes I had to make to trigger the issue are the following: 

1. The connection should be "authenticated", and not just "bonded" (else the issue apparently isn't triggered). By enabling "DISPLAY YESNO" IO capabilities on both devices, an authenticated bond is made. 
2. My sm key config differs from the example:
    `ble_hs_cfg.sm_our_key_dist = BLE_SM_PAIR_KEY_DIST_ENC | BLE_SM_PAIR_KEY_DIST_ID;`
    `ble_hs_cfg.sm_their_key_dist = BLE_SM_PAIR_KEY_DIST_ENC | BLE_SM_PAIR_KEY_DIST_ID;`
 
Afterwards the issue can be reproduced by following the steps in my original issue description. I did need to disconnect the peripheral (by power cycling it) one extra time between steps 4 and 5 before seeing the address type change. Make sure to start from devices that don't have a bond yet (erase-flash), and to test with 2 ESP32-S3 devices.

I tested this on idf v5.2.1, but in the meantime I've also tested on idf v5.3 and the issue still persists.
 
The modified examples, as well as the logs while running my test:
[blecent-bleprph-IDFGH-12413.zip](https://github.com/user-attachments/files/16661704/blecent-bleprph-IDFGH-12413.zip)
[blecent_output.txt](https://github.com/user-attachments/files/16661705/blecent_output.txt)
[bleprph_output.txt](https://github.com/user-attachments/files/16661706/bleprph_output.txt)