espressif/esp-idf #13438: ESP_ERR_MESH_NO_MEMORY

Leadrive, 2024-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.6

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

不出现ESP_ERR_MESH_NO_MEMORY

### What is the actual behavior?

之前也有人反馈此问题，但没有人解决。https://github.com/espressif/esp-mdf/issues/315
症状是这样的，esp_mesh_send，如果返回ESP_ERR_MESH_TIMEOUT，大约持续10几次，就会开始不断出现ESP_ERR_MESH_NO_MEMORY，然后就一直无法正常运行了，此时堆内存这些都是有的，推测就是mesh库没把发送失败的清除，导致积压。
从应用角度来看，此时并没有收到MESH_EVENT_CHILD_DISCONNECTED, MESH_EVENT_PARENT_DISCONNECTED之类的断开消息，肯定会认为mesh是正常的，还是会不断发送。
root端的日志，看起来很正常，没报任何异常

node端的日志
````
I (537592) [app_main, 43]: Receive binary data, cmd:1, num:235, size:1200
I (537598) [handle_mesh_update, 123]: Flash write success, dataLen:1200, num:235, UpdateTotalNum:1611
I (542126) [mesh_netif, 225]: Sending to root, dest addr: ff:ff:ff:ff:ff:ff, size: 42
E (542126) [mesh_netif, 226]: Send with err code 16394 ESP_ERR_MESH_TIMEOUT
E (542514) [handle_read_data, 75]: HandleReadData, msgId:643, cmdId:29 already timeout:1 s
I (542539) [mesh_node, 150]: NodeWriteExec, now:1711008757, data: xxxx                                                                                                                                                                                               
I (543113) [app_main, 43]: Receive binary data, cmd:1, num:236, size:1200
I (543119) [handle_mesh_update, 123]: Flash write success, dataLen:1200, num:236, UpdateTotalNum:1611
E (546100) [handle_read_data, 75]: HandleReadData, msgId:645, cmdId:29 already timeout:1 s
W (547492) mesh: [mesh_schedule.c,3131] [WND-RX]max_wnd:9, 1800 ms timeout, seqno:0, xseqno:433, no_wnd_count:0, timeout_count:0
E (548684) [handle_read_data, 75]: HandleReadData, msgId:646, cmdId:29 already timeout:1 s
E (548696) [mesh_node, 127]: NodeWriteExec, timeout, send fail message, msgId:647, cmdId:101, endTime:1711008762, now:1711008763
E (548700) [handle_message, 139]: HandleCommonContent, split error, content=, num=1
I (559026) [mesh_node, 150]: NodeWriteExec, now:1711008773, data: xxxx                                                                                                                                                                                                          
I (559934) [app_main, 43]: Receive binary data, cmd:1, num:237, size:1200
I (559939) [handle_mesh_update, 123]: Flash write success, dataLen:1200, num:237, UpdateTotalNum:1611
I (559979) [mesh_node, 150]: NodeWriteExec, now:1711008774, data: xxxxx                                                                                                                                                                                                                        
I (561183) [app_main, 43]: Receive binary data, cmd:1, num:238, size:1200
I (561188) [handle_mesh_update, 123]: Flash write success, dataLen:1200, num:238, UpdateTotalNum:1611
I (561191) [mesh_node, 150]: NodeWriteExec, now:1711008775, data: xxxxx 
W (566768) [mesh_internal, 34]: MeshSendData, err=<ESP_ERR_MESH_TIMEOUT>
E (566768) [mesh_node, 162]: <ESP_ERR_MESH_TIMEOUT> nodeWriteTask, msgId:656, cmdId:101
W (571874) [mesh_internal, 34]: MeshSendData, err=<ESP_ERR_MESH_TIMEOUT>
[... 100 lines cut ...]
E (693275) [mesh_node, 162]: <ESP_ERR_MESH_NO_MEMORY> nodeWriteTask, msgId:700, cmdId:29
W (696277) [mesh_internal, 32]: MeshSendData, err=<ESP_ERR_MESH_NO_MEMORY>, 内部堆=64551, 外部堆=2022727
E (696278) [mesh_node, 162]: <ESP_ERR_MESH_NO_MEMORY> nodeWriteTask, msgId:701, cmdId:29
W (699279) [mesh_internal, 32]: MeshSendData, err=<ESP_ERR_MESH_NO_MEMORY>, 内部堆=65271, 外部堆=2023447
E (699280) [mesh_node, 162]: <ESP_ERR_MESH_NO_MEMORY> nodeWriteTask, msgId:702, cmdId:29
W (702389) [mesh_internal, 32]: MeshSendData, err=<ESP_ERR_MESH_NO_MEMORY>, 内部堆=63463, 外部堆=2021639
E (702390) [mesh_node, 162]: <ESP_ERR_MESH_NO_MEMORY> nodeWriteTask, msgId:703, cmdId:29
W (705380) [mesh_internal, 32]: MeshSendData, err=<ESP_ERR_MESH_NO_MEMORY>, 内部堆=63463, 外部堆=2021639
E (705380) [mesh_node, 162]: <ESP_ERR_MESH_NO_MEMORY> nodeWriteTask, msgId:704, cmdId:29
W (708370) [mesh_internal, 32]: MeshSendData, err=<ESP_ERR_MESH_NO_MEMORY>, 内部堆=63463, 外部堆=2021639
E (708370) [mesh_node, 162]: <ESP_ERR_MESH_NO_MEMORY> nodeWriteTask, msgId:705, cmdId:29
W (711373) [mesh_internal, 32]: MeshSendData, err=<ESP_ERR_MESH_NO_MEMORY>, 内部堆=63463, 外部堆=2021639
E (711374) [mesh_node, 162]: <ESP_ERR_MESH_NO_MEMORY> nodeWriteTask, msgId:706, cmdId:29
W (714003) [mesh_internal, 32]: MeshSendData, err=<ESP_ERR_MESH_NO_MEMORY>, 内部堆=63463, 外部堆=2021639
E (714004) [mesh_node, 162]: <ESP_ERR_MESH_NO_MEMORY> nodeWriteTask, msgId:707, cmdId:29
W (716993) [mesh_internal, 32]: MeshSendData, err=<ESP_ERR_MESH_NO_MEMORY>, 内部堆=63463, 外部堆=2021639
E (716993) [mesh_node, 162]: <ESP_ERR_MESH_NO_MEMORY> nodeWriteTask, msgId:708, cmdId:29
W (719989) [mesh_internal, 32]: MeshSendData, err=<ESP_ERR_MESH_NO_MEMORY>, 内部堆=63463, 外部堆=2021639
E (719989) [mesh_node, 162]: <ESP_ERR_MESH_NO_MEMORY> nodeWriteTask, msgId:709, cmdId:29
I (721440) [mesh_netif, 225]: Sending to root, dest addr: ff:ff:ff:ff:ff:ff, size: 42
````

### Steps to reproduce.

故障重现，可能人为在root节点连接的路由器上，制造网络连接不顺，或时断时续，可能容易复现此错误。

Leadrive, 2024-03-21:
建议你们买一个这路由器TY-6201A试下，中国电信送的，闲鱼几十元 一个，这路由器跟esp32相克
[image]

Leadrive, 2024-03-22:
esp_wifi_set_ps设置为省电模式，比较容易复现此问题，省电时，经常超时。

mhdong, 2024-03-25:
Hi @Leadrive 
同款路由器已购买，后面我们会尝试本地复现问题并解决。

AxelLin, 2024-06-06:
> Hi @Leadrive 同款路由器已购买，后面我们会尝试本地复现问题并解决。

@mhdong Any update?

zhangyanjiaoesp, 2024-06-06:
@Leadrive Can you provide a demo to test?

MosamXu, 2024-06-11:
ESP32-S3 V5.2.2 也有同样的问题

zhangyanjiaoesp, 2024-06-12:
@MosamXu Can you provide a demo to test?

Sherry616, 2024-08-09:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates. 
If you meet a similar issue, please feel free to start a new discussion in the forum.
Thanks for using our Espressif product!

closed 2024-08-09 by Sherry616