espressif/esp-idf #13439: Accidentally writing to file system when printing to stdout with ESP_LOGx()

voedipus, 2024-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Steps to reproduce:
1. Select "None" as channel for console output 
[image]
2. Open a file in file system with **fopen()** and see that it is opened with fileno = 0 (with debugging)
3. Call fopen() once again(to open stdout with fileno = 1)
4. Print something to stdout with ESP_LOGx()
5. Read second opened file and see that ESP_LOGx() write there

If channel for console output is UART0 then fopen() opens first file with fileno = 3

igrr, 2024-03-21:
@voedipus Would you mind posting the code you used in these two steps?

> Open a file in file system with fopen() and see that it is opened with fileno = 0 (with debugging)
> Call fopen() once again(to open stdout with fileno = 1)

Could you also mention the version of ESP-IDF you are using?

voedipus, 2024-03-21:
@igrr I used example project named "file_serving" with minor change in app_main:
```c
void app_main(void)
{
    ESP_LOGI(TAG, "Starting example");
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    /* Initialize file storage */
    const char* base_path = "/data";
    ESP_ERROR_CHECK(example_mount_storage(base_path));

    /* This helper function configures Wi-Fi or Ethernet, as selected in menuconfig.
     * Read "Establishing Wi-Fi or Ethernet Connection" section in
     * examples/protocols/README.md for more information about this function.
     */
    ESP_ERROR_CHECK(example_connect());

    /* Start the file server */
    ESP_ERROR_CHECK(example_start_file_server(base_path));
    ESP_LOGI(TAG, "File server started");
    FILE* first = NULL;
    first = fopen("/data/test1.txt", "w");
    FILE* second = NULL;
    second = fopen("/data/test2.txt", "w");

    while (1){
        ESP_LOGE(TAG, "Is this written to FS?");
        vTaskDelay(200);
    }
}
```

As you can see I open 2 files(test1.txt and test2.txt) and write to stdout using ESP_LOGE()
[image]

There were no explicit writes to test2.txt, but it's size is more than 0

Contents of the test2.txt file
[image]
```

haberturdeur, 2024-04-03:
The issue seems to be that vfs doesn't reserve file descriptors 0 (STDIN), 1(STDOUT) and 2(STDERR):
So when the second file gets open it gets fd 1. Then vprintf in ESP_LOGX assumes that fd 1 is STDOUT, and writes to it...
[image]

closed 2024-05-30 by espressif-bot