espressif/esp-idf #13440: Typo in Inter-Integrated Circuit documentation

nopnop2002, 2024-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

This example is described on the following page, but there is a typo.
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/i2c.html

```
out_voltage = Vref * digi_val / 255

i2c_device_config_t dev_cfg = {
    .dev_addr_length = I2C_ADDR_BIT_LEN_7,
    .device_address = 0x58,
    .scl_speed_hz = 100000,
};

i2c_master_dev_handle_t dev_handle;
ESP_ERROR_CHECK(i2c_master_bus_add_device(I2C_PORT_NUM_0, &dev_cfg, &dev_handle));
uint8_t buf[20] = {0x20};
uint8_t buffer[2];
//ESP_ERROR_CHECK(i2c_master_transmit_receive(i2c_bus_handle, buf, sizeof(buf), buffer, 2, -1));
ESP_ERROR_CHECK(i2c_master_transmit_receive(dev_handle, buf, sizeof(buf), buffer, 2, -1));
```

master(latest) and documents other than ESP32 are the same.

mythbuster5, 2024-04-11:
Fixed

closed 2024-04-11 by mythbuster5

grginho, 2024-04-26:
Hello,

Sorry if this is not directly related to this issue, but I have a question related to the I2C documentation.

In the Documentation for the new I2C driver, it says this:
Thread Safety
The factory function i2c_new_master_bus() and i2c_new_slave_device() are guaranteed to be thread safe by the driver, which means, user can call them from different RTOS tasks without protection by extra locks. Other public I2C APIs are not thread safe. which means the user should avoid calling them from multiple tasks, if user strongly needs to call them in multiple tasks, please add extra lock.

Does this mean that only i2c_new_master_bus() and i2c_new_slave_device() are thread safe, and all other I2C functions are not thread safe?

Regards,

Dragan

nopnop2002, 2024-04-26:
@grginho 

You are right.

mythbuster5, 2025-02-06:
@grginho See latest document, some changes applied there.