espressif/esp-idf #13441: using esp_wifi_scan_start before esp_wifi_connect changes behavior of connect

pggh, 2024-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

own

### Power Supply used.

USB

### What is the expected behavior?

esp_wifi_connect should always behave the same, regardless what scan has been performed in the past.

### What is the actual behavior?

When esp_wifi_scan_start has been used (and completed) to do an active scan, subsequent esp_wifi_connect will use the timing from the previous scan.

### Steps to reproduce.

1. do an active scan with .scan_time.active.max = 1
2. call esp_wifi_connect for an none existent AP; connect fails after 20ms
3. do an active scan with .scan_time.active.max = 1000
4. call esp_wifi_connect for an none existent AP; connect fails after 13010ms

Country-Setting set to
      wifi_country_t c = {
         .cc = "DE",
         .schan = 1,
         .nchan = 13,
         .max_tx_power = 20,
         .policy = WIFI_COUNTRY_POLICY_MANUAL
      };
      ESP_ERROR_CHECK(esp_wifi_set_country(&c));
to allow active scan on all channels.

### More Information.

From the observed behavior, it looks like the scan configuration is stored in a global variable and the later also used for the scan performed by esp_wifi_connect without setting the parameters for the connect.

mhdong, 2024-03-22:
Hi @pggh 

> From the observed behavior, it looks like the scan configuration is stored in a global variable and the later also used for the scan performed by esp_wifi_connect without setting the parameters for the connect.
You are right.
https://github.com/espressif/esp-idf/blob/be06a6f5ffe36f9554cfc91fe2036e0fc85fea60/components/esp_wifi/include/esp_wifi.h#L454
esp_wifi_scan_start will config settings for scan.

pggh, 2024-03-22:
Actually it would be pretty useful to have an API to control the timing of esp_wifi_connect:
- setting the max_time
- limiting connect to a single channel

For battery powered devices, this could greatly reduce power consumption.
On wake-up from deep-sleep, the ESP could try to connect to the last used AP using a short max_time like 10ms on the last used channel two or three times and go to sleep after about 30ms when not successful.

TD-er, 2024-03-22:
When using WiFi 6 (802.11ax) you can negotiate with the AP to remain connected for some time during sleep.

When (re)connecting to an AP, you can specify a specific channel and BSSID.
Not sure how it is done in esp-IDF code, but you might want to look at how it is done in the Arduino code and which IDF functions/structs they use as Arduino's `WiFi.begin()` does have such options.
I doubt you can connect using a timeout less than 100 msec as that's the typical interval period of an AP to respond.
No idea if you can specify such a timeout, just stating that it would be rather useless to have it set to less than 100 msec.

pggh, 2024-03-22:
@TD-er 
- ESP32-S3 does not support WiFi 6 as far as I know.
- The channel specified does not limit the connect-scan to that channel. It only changes the order in which channels are scanned. It documented and working that way.
- It does make sense to use a timeout <100ms. The AP answers the request immediately (when channel not congested). The bug from this issue let me test this.

mhdong, 2024-03-28:
Hi @pggh 
We will add some new api to solve this issue. After the code merged, it will be updated here.

mhdong, 2024-04-24:
Hi @pggh 
Issue solved with https://github.com/espressif/esp-idf/commit/770162d76cee65f8b2c1e4701d085f89acef20ef

Sherry616, 2024-04-24:
Thanks for reporting, we will close this issue. Feel free to reopen, thanks.

closed 2024-04-24 by Sherry616

AxelLin, 2024-04-24:
> Hi @pggh Issue solved with [770162d](https://github.com/espressif/esp-idf/commit/770162d76cee65f8b2c1e4701d085f89acef20ef)

Just to clearify if my understanding is correct:
So you expect users to call esp_wifi_get_scan_parameters() before using esp_wifi_scan_start(), then calling esp_wifi_set_scan_parameters() before esp_wifi_connect() ?

mhdong, 2024-04-24:
Hi @AxelLin 
- esp_wifi_set_scan_parameters() : Used to set the default scan parameters to be used
while scanning. These parameters will be used for all scans conducted by station
including connect scan. Parameters provided through esp_wifi_scan_start() override the
default parameters for that scan only
- esp_wifi_get_scan_parameters(): Used to get the current default scan parameters.
You do not have to call esp_wifi_set_scan_parameters() before esp_wifi_connect().

pggh, 2024-04-29:
It look like the new API does not support limiting the channels on which connect attempts are made. (see original comment from Mar 22).
The scan API now has a channel-bitmap which is useful. I wish this or limiting to a single channel would be available for connect too. This could speed up connects in fail case.