espressif/esp-idf #13443: Ethernet driver deinitialization woes

mickeyl, 2024-03-21:
Following up on a message by @kostaond in https://github.com/espressif/esp-idf/issues/5697#issuecomment-2011471354_ where we discussed a missing deinit for the bridge example, I want to help getting this fixed.

To simplify things, I first want to get this working without the bridge functionality. Inversing the following init:

```c
    // Create Ethernet interface with appropriate configuration
    auto netif = esp_netif_new(&cfg);

    // Attach Ethernet driver to TCP/IP stack
    ESP_ERROR_CHECK(esp_netif_attach(netif, esp_eth_new_netif_glue(eth_handles[0])));

    // Register user defined event handers
    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &got_ip_event_handler, NULL));

    // Start Ethernet driver state machine
    for (int i = 0; i < eth_port_cnt; i++) {
        ESP_ERROR_CHECK(esp_eth_start(eth_handles[i]));
    }
```

I figured I'd start w/ 

```c
    // Stop Ethernet driver state machine
    for (int i = 0; i < eth_port_cnt; i++) {
        ESP_ERROR_CHECK(esp_eth_stop(eth_handles[i]));
    }

    // Unregister user defined event handlers
    esp_event_handler_unregister(ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler);
    esp_event_handler_unregister(IP_EVENT, IP_EVENT_ETH_GOT_IP, &got_ip_event_handler);

    // Delete Ethernet driver glue
    esp_eth_del_netif_glue((esp_eth_netif_glue_handle_t)netif->driver_handle);

    // Destroy Ethernet interface
    esp_netif_destroy(netif);

    // Destroy Ethernet drivers (technically part of the glue, but the glue is a private struct)
    example_eth_deinit(eth_handles, eth_port_cnt);
}
```

But this crashes in esp_eth_decrease like that:

[image]

mickeyl, 2024-03-21:
Debugging more into this, it seems I can't get the glue reference via `(esp_eth_netif_glue_handle_t)netif->driver_handle)`, because in `esp_netif_attach` the `driver_handle` seems to get rewritten by the means of a `post_attach` callback.

mickeyl, 2024-03-21:
Augmenting the deinit sequence in `esp_eth.c` with
```c
    ESP_GOTO_ON_ERROR(phy->deinit(phy), err, TAG, "deinit phy failed");
    ESP_GOTO_ON_ERROR(phy->del(phy), err, TAG, "del phy failed"); // new
    ESP_GOTO_ON_ERROR(mac->deinit(mac), err, TAG, "deinit mac failed");
    ESP_GOTO_ON_ERROR(mac->del(mac), err, TAG, "del mac failed"); // new
```

Together with the following new function in the `ethernet_init` component:

```c
void example_eth_deinit(esp_eth_handle_t *eth_handles, uint8_t eth_cnt) {

    for (int i = 0; i < eth_cnt; i++) {
        esp_eth_driver_uninstall(eth_handles[i]);
    }
    free(eth_handles);

#ifdef CONFIG_EXAMPLE_USE_SPI_ETHERNET
    spi_bus_free(CONFIG_EXAMPLE_ETH_SPI_HOST);
#endif
}

seems to be a step in the right direction. I can bring up / bring down Ethernet a couple of times, before it crashes somewhere.
```

mickeyl, 2024-03-21:
Ok, stopping the DHCP server manually before stopping the state machine on the drivers seems to be the missing link. Before tackling the more complicated bridge example, I'd like to ask how we can proceed from here. It looks like there is little missing to make this work reliably, in particular calling `del` in the `phy` and `mac` and adding the aforementioned `example_eth_deinit` in the component.

@kostaond or anyone else: What do you think?

kostaond, 2024-03-22:
> Is there a special reason why the esp_eth_driver_uninstall does not already call the corresponding del functions?

Yes, `mac` and `phy` were created (and resources allocated) by separated function call, not by `esp_eth_driver_install`. Therefore it's not job of `esp_eth_driver_uninstall`.

I'm currently in process of Ethernet driver update. The current issue with using `example_eth_init` is that you (user) lost reference to `mac` and `phy` instances and so you cannot delete them. This can be easily overcome as demonstrated by [Ethernet init Component](https://github.com/espressif/esp-eth-drivers/blob/a631cfe5f74ad677462bee880566d85d55f08427/ethernet_init/ethernet_init.c#L72). However, I don't like that solution much when considering `eth_hanler` keeps the reference to both instances. Hence, I'm planning to update `esp_eth` driver to provide getter functions for these two instances. Along with `esp_eth` update, I'll update deinit sequence for both Ethernet init example/component.

mickeyl, 2024-03-22:
Very good, thanks! I'll keep the aforementioned diff in my fork for now, but will wait for a proper solution then.

kostaond, 2024-03-25:
@KaeLL would https://github.com/espressif/esp-eth-drivers/pull/21 help you solve your problem? The original intention of the PR was different, but it's `Dummy PHY` and it sounds exactly you need. If it helps, I'll update the README with your use case :)

KaeLL, 2024-03-27:
I skimmed over it and seems like that is what I was looking for. I'll take a deeper look the next time I update my code.

mickeyl, 2025-01-23:
I've seen https://github.com/espressif/esp-idf/commit/d30e4d9960afd0d1ed067fef741126af810f675e landing. @kostaond Is this helping for our deinitialization issue?

kostaond, 2025-01-24:
@mickeyl, the updated you referenced to is basically "non-functional" (from higher perspective). It's been just about replacing GPIO functions with functions which does not overwrite all the current IO configurations. Ethernet de-init can be found at https://github.com/espressif/esp-idf/blob/master/examples/ethernet/basic/components/ethernet_init/ethernet_init.c#L309

cederom, 2026-02-03:
+1 for adding documentation and examples on how to properly and safely deinit eth this is missing in v5.5.2 :-)