espressif/esp-idf #13445: TCP socket Software caused connection abort (113)

dizcza, 2024-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1 tag 9f7eacb089a1ae16da858f55f41c4f15b4c0f137

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

TTGO T8 v1.7

### Power Supply used.

USB

### What is the expected behavior?

TCP socket connected

### What is the actual behavior?

TCP socket unable to connect: errno Software caused connection abort (113)

### Steps to reproduce.

```c
#define TCP_SOCK_INVALID              (-1)
#define TCP_CLIENT_TIMEOUT_SEC  (10)

static int m_sock = TCP_SOCK_INVALID;

static const char *TAG = "tcp_client";

esp_err_t tcp_client_connect(const char *host, int port) {  // port 8333
    int addr_family = 0;
    int ip_protocol = 0;

    struct sockaddr_in dest_addr;
    inet_pton(AF_INET, host, &dest_addr.sin_addr);
    dest_addr.sin_family = AF_INET;
    dest_addr.sin_port = htons(port);
    addr_family = AF_INET;
    ip_protocol = IPPROTO_IP;

    m_sock = socket(addr_family, SOCK_STREAM, ip_protocol);
    if (m_sock < 0) {
        ESP_LOGW(TAG, "Unable to create socket: errno %s (%d)", strerror(errno), errno);
        return ESP_FAIL;
    }
    int opt = 1;
    setsockopt(m_sock, SOL_SOCKET, SO_REUSEADDR, &opt, sizeof(opt));

    struct timeval tv;
    tv.tv_sec = TCP_CLIENT_TIMEOUT_SEC;
    tv.tv_usec = 0;
    setsockopt(m_sock, SOL_SOCKET, SO_RCVTIMEO, (const char*)&tv, sizeof tv);

    ESP_LOGI(TAG, "Socket created, connecting to %s:%d ...", host, port);
    int err_tcp = connect(m_sock, (struct sockaddr *)&dest_addr, sizeof(dest_addr));
    esp_err_t err;
    if (err_tcp != 0) {
        err = ESP_FAIL;
        ESP_LOGW(TAG, "TCP socket unable to connect: errno %s (%d)", strerror(errno), errno);
        tcp_client_close();
    } else {
        err = ESP_OK;
        ESP_LOGI(TAG, "Successfully connected to %s:%d", host, port);
    }

    return err;
}

void tcp_client_close() {
    if (m_sock >= 0) {
        shutdown(m_sock, 0);
        close(m_sock);
        m_sock = TCP_SOCK_INVALID;
    }
}
```

### Debug Logs.

```plain
Socket created, connecting to xxxx:8333
(after 10 sec)
TCP socket unable to connect: errno Software caused connection abort (113)
```

### More Information.

A Python TCP server is set up and running on vps.ag, it has been running for years without a problem. I have several ESP32 boards periodically pinging this server and doing recv/send. The last successful transaction was on 2024.03.21 01:44:31. ASAIK, vps.ag didn't have any maintenance at that time. And currently, I can send and receive data using a simple python TCP client script:

```python
import socket

host = xxxxxx
port = 8333
s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.connect((host, port))
s.sendall(b'Hello, world')
data = s.recv(1024)
s.close()
print('Received', repr(data))
```

So it's hardly a problem with the TCP server.

https://github.com/espressif/arduino-esp32/issues/6580 could be related.

---

The communication with ESP32 boards relies on this TCP connection. I cannot communicate with my boards any longer - it's crucial.

dizcza, 2024-03-21:
Of course, I tried erasing the flash of a sample ESP32 board which I can access locally.

Launching the TCP server on a different port, say 8334, doesn't help either.

Launching the TCP server on my host desktop and re-hardcoding the server IP to my local home IP does help.

hansw123, 2024-03-22:
hi @dizcza 
Thank you for your feedback. I can't find out the exact cause without more information, could you provide a wifi capture and a tcp capture?

dizcza, 2024-03-22:
Any chance to capture a TCP packet from within the ESP32 itself? I'm having trouble capturing packets on a protected network using the monitor mode... Well, maybe I can set up an open AP for a test.

dizcza, 2024-03-22:
While I was studying how to decrypt packets on an encrypted AP with a network adapter in the monitor mode and Wireshark, all this jazz, the communication was restored. Now ESP32 boards can reach my TCP server. It's strange because I haven't done anything to either the TCP server or ESP boards...
The issue will probably occur again. Till that point, I'm closing it.

closed 2024-03-22 by dizcza

hansw123, 2024-03-25:
@dizcza 
Thank you for your help，feel free to reopen