espressif/esp-idf #13447: Console component won't build for linux targets but doesn't explain why

bdoherty-tovala, 2024-03-21:
https://github.com/espressif/esp-idf/blob/cf7b9bed4c57e0f42a109f9417f04ba799e41d0b/components/console/CMakeLists.txt#L4

In v5.2, if you try to build a component for host-based linux testing that depends on console (in my case, via console_simple_init,) the build will mysteriously fail due to being unable to include `esp_console.h` 

It took me quite a long time of digging a head-scratching to find this line that explains why.

Even though this capability appears possible in the master branch, it is not in any official release. It'd be super sweet if it were in an official release, but at the very least, trying to depend on this component for a linux target build when it's not supported should complain loudly that it's explicitly not supported, instead of failing mysteriously.

KonstantinKondrashov, 2024-03-22:
Hi @bdoherty-tovala!
We'll see how we can improve it, to make better experience with it.
We currently use this silent return to enable builds without `set(COMPONENTS main)` .
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/host-apps.html?highlight=linux#requirements-for-using-mocks
Thanks.

fhrbata, 2024-05-31:
Hello @bdoherty-tovala ,

thank you for your report! It revealed a problem in the build system, specifically with a component dependency check that is not functioning as intended. The proper fix changes the build system's behavior, which could cause unexpected issues for users of stable versions. Considering this and the fact that it pertains to the Linux target, which is still in preview, we do not plan to backport the fix to stable versions. Instead, we will address it only in the master (v5.4) branch.

Thank you very much!

closed 2024-06-01 by espressif-bot