espressif/esp-idf #13448: Please implement the ShockBurst protocol

kriegste, 2024-03-21:
### Is your feature request related to a problem?

The ShockBurst protocol uses the 2.4G band and has the same modulation as BLE. Yet, to use it we still require a separate chip! It should be a no-brainer for Espressif to implement this. Packet structure is simple- it is explained here:
https://devzone.nordicsemi.com/nordic/nordic-blog/b/blog/posts/intro-to-shockburstenhanced-shockburst
There is an enhanced version... the Enhanced ShockBurst protocol. Of course it would be optimal to implement this, too. But if you have legal concerns, you could just implement the legacy version and let us users deal with the differences it in our code. The original legacy ShockBurst protocol is 20 years old!

zenghi, 2024-04-16:
@zhp0406 可以帮忙加上这个功能么，我也有这个需求，单纯使用BLE数据格式作为简单的2.4G无线收发功能  
[image]
esp_phy_ble_rx  esp_phy_ble_tx  使用这2个接口可以正常通信，但是没办法实时获取接收的数据内容和自定义发送的数据内容。并且想着能够兼容正在跑的程序。

zhp0406, 2024-04-18:
@zenghi 
我们正在进行评估,并尽快给你回复.

zhp0406, 2024-04-19:
@kriegste 
We will evaluate the feature and get back to you as soon as possible.

zhp0406, 2024-04-19:
@zenghi @kriegste 
Please contact the business.
https://www.espressif.com/zh-hans/contact-us/sales-questions

jinnie11, 2024-04-26:
> @zhp0406 这边已经联系商务，目前感觉esp-idf加这个功能应该是很轻松，因为esp_phy_ble_rx esp_phy_ble_tx 就再这两个上面做个改动，目前想要类似eps-now的功能 但是是使用2.4G的BLE格式进行通信，主要解决超快唤醒低功耗等一些其他应用问题。
> 
> 下面是沁恒微的说法 ![image](https://private-user-images.githubusercontent.com/6140297/324768688-f4f4e5bd-59b9-48af-8441-4beb82b43f66.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQxMzM0NzEsIm5iZiI6MTcxNDEzMzE3MSwicGF0aCI6Ii82MTQwMjk3LzMyNDc2ODY4OC1mNGY0ZTViZC01OWI5LTQ4YWYtODQ0MS00YmViODJiNDNmNjYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MjZUMTIwNjExWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OGNjNDc2NWFkNTFmODMyZDFmNTMxYTQwNDljZDU5NjhkNWVmZjI4M2Y3YzA0ZDM0MWI3NjZjYTQzOWFjMjllNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.oG0S-Y54Ew0faBMAdmo9m7kkRdrL45qyihtk-268sgY)

您留的邮箱可以告知下，我们看下对接

reucsan, 2025-02-07:
Hello @kriegste,

Have you found any solution for using ShockBurst / Enhanced ShockBurst with ESP32?

kriegste, 2025-02-07:
Hi. No, Espressif is unwillig to implement it. So using a separate chip is the only solution.

esp-zhp, 2025-02-08:
Please contact the business team for this feature request. Thank you.