espressif/esp-idf #13449: Deep Sleep Enable GPIO Wakeup() won't enable on S3

SolidStateLEDLighting, 2024-03-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I'm following the IDF example in how to work with Deep Sleep.   I am not able to use this function:

esp_deep_sleep_enable_gpio_wakeup(BIT(DEFAULT_WAKEUP_PIN), DEFAULT_WAKEUP_LEVEL)

The include/source files require a definition of SOC_GPIO_SUPPORT_DEEPSLEEP_WAKEUP = 1

Compiler can't find this **SOC_GPIO_SUPPORT_DEEPSLEEP_WAKEUP** definition for some reason.

I have looked in header files, sdkconfig, Kconfig -- I just don't know where else to look to enable this setting.

Sometimes intellisense can see this setting in esp32p4\include\soc and sometimes it sees it in esp32c2\include\soc -- but this not correct because I'm setup for s3.

The setting and the functions are related to the component esp_hw_support and I have added that to my Requires CMakeLists file and this doesn't help at all.   There is a good chance that esp_hw_support is already a component in my project.

### What is the actual behavior?

C:/Users/<User>/Desktop/ESP_Projects/v.5.2/project-name/main/main_sleep.cpp:24:5: error: 'esp_deep_sleep_enable_gpio_wakeup' was not declared in this scope; did you mean 'esp_sleep_enable_gpio_wakeup'?

### Steps to reproduce.

Just try to use the function:

esp_deep_sleep_enable_gpio_wakeup(BIT(DEFAULT_WAKEUP_PIN), DEFAULT_WAKEUP_LEVEL)

esp-wzh, 2024-03-22:
Hi, @SolidStateLEDLighting 

The hardware of esp32s3 does not support waking up deepsleep through GPIO. The corresponding function can be replaced by ext0/ext1 wake-up source. Please refer to: 

https://docs.espressif.com/projects/esp-idf/en/release-v5.2/esp32s3/api-reference/system/sleep_modes.html#external-wakeup-ext0

https://docs.espressif.com/projects/esp-idf/en/release-v5.2/esp32s3/api-reference/system/sleep_modes.html#external-wakeup-ext1

SolidStateLEDLighting, 2024-03-22:
Oh... crap.    No wonder it didn't make sense to me.    Sorry to bother you on this... and I appreciate the guidance.

I just don't know why I didn't get this understanding from the documentation.

SolidStateLEDLighting, 2024-03-22:
Give me a day, I'll close out this ticket.

SolidStateLEDLighting, 2024-03-22:
Confirmed, the correct method for using GPIO (RTC pin) to wake from a deep sleep is to configure that input as an ext0 wakeup.

closed 2024-03-22 by SolidStateLEDLighting