espressif/esp-idf #13450: BT BTc: bte_blufi_recv handler payload is longer than packet length, len 252

Leadrive, 2024-03-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.6

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

blufi通讯正常

### What is the actual behavior?

BT BTc: bte_blufi_recv handler payload is longer than packet length, len 252
有时候不定时会出现此消息，太难复现了，都不知道如何调试定位。

### Steps to reproduce.

使用微信小程序，通过blufi配网，有时候会出现通讯包太长。
感觉是加密后，传输的包过长。

有没有什么办法，取消加密模式？

### Debug Logs.

```plain
`
[10:26:53.141]收+◆口[0:31mE (31022) BT BTc: bte_blufi_recv handler payload is longer than packet length, len 252
(29)[BLUFI,335]:event:25□[0m:32m10:31mE(31025)BLUFI:BLUFI report error,error code 9
````
不断重复出现此错误，然后无法正常通讯
```

Leadrive, 2024-03-22:
````
�[0;31mE (32342) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (32402) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (32462) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (32522) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (32582) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (32642) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (32702) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (32762) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (32852) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (32912) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (32972) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (33062) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (33122) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (33182) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (33242) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (33302) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (33362) BT_BTC: btc_blufi_recv_handler payload is longer than packet length, len 264 
�[0m
�[0;31mE (33422) BT_BTC�[0;32mI (32314) [handle_message, 154]
````

Leadrive, 2024-03-22:
交换密钥，有时候也失败，并且要消耗好几秒钟。
通讯不想加密，只想稳定、快速连接，有什么办法取消加密模式？

zhp0406, 2024-03-26:
@Leadrive 
微信小程序你使用的哪个？加密是微信小程序端发起的,我需要确定一下微信小程序的行为。
手机端也有个 blufi app，可以选择加密或者不加密。

Leadrive, 2024-03-26:
@zhp0406 谢谢回复。
小程序用的 https://github.com/xuhongv/BlufiEsp32WeChat
作者说是从app中改过来的，麻烦帮看看。

zhp0406, 2024-03-26:
@Leadrive 
第三方的小程序我们并不了解，你可以尝试在源码中找到发起加密的地方并屏蔽主动发起加密。

closed 2024-03-26 by Leadrive