espressif/esp-idf #13453: xtensa-esp32-elf-gcc Problems

Jimmy106-e, 2024-03-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Operating System used.

macOS

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It should flash the app to my ESP32 Cam

### What is the actual behavior?

it returns this error ```xtensa-esp32-elf-gcc: error: unrecognized command line option '-mfix-esp32-psram-cache-issue'```

### Steps to reproduce.

1. export IDF_PATH=~/esp32/esp-idf
2. export PATH=$HOME/esp32/xtensa-esp32-elf/bin:$PATH
3. make flash monitor

### Build or installation Logs.

```plain
Toolchain path: /Users/DGB/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Compiler version: 5.2.0
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/micro-ecc/micro-ecc/uECC.o
AR build/bootloader/micro-ecc/libmicro-ecc.a
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/soc/esp32/gpio_periph.o
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/esp32/rtc_clk_init.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/rtc_periph.o
CC build/bootloader/soc/esp32/rtc_pm.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/rtc_time.o
CC build/bootloader/soc/esp32/rtc_wdt.o
CC build/bootloader/soc/esp32/sdio_slave_periph.o
CC build/bootloader/soc/esp32/sdmmc_periph.o
CC build/bootloader/soc/esp32/soc_memory_layout.o
CC build/bootloader/soc/esp32/spi_periph.o
CC build/bootloader/soc/src//memory_layout_utils.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
LD build/bootloader/bootloader.elf
esptool.py v2.6
Python requirements from /Users/DGB/esp32/esp-idf/requirements.txt are satisfied.
Building partitions from /Users/DGB/Desktop/esp32-homekit-camera/partitions.csv...
CC build/app_trace/app_trace.o
xtensa-esp32-elf-gcc: error: unrecognized command line option '-mfix-esp32-psram-cache-issue'
make[1]: *** [app_trace.o] Error 1
make: *** [component-app_trace-build] Error 2
```

igrr, 2024-03-22:
You seem to be using what seems to be a very old version of the toolchain:
```
WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
```
and the toolchain version `crosstool-ng-1.22.0-61-gab8375a` did not support the `-mfix-esp32-psram-cache-issue` argument.

---

I think your IDF version also can't be v5.2.1, since support for GNU Make based build (`make flash monitor`) was removed in v5.0. Judging by the "Expected to see version" line, your IDF version is [not later than v3.3.2](https://github.com/espressif/esp-idf/blob/v3.3.2/tools/toolchain_versions.mk).

You might be following some old tutorial, I'm guessing. I would recommend reinstalling your environment following the getting started guide: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/index.html

Jimmy106-e, 2024-03-22:
> You seem to be using what seems to be a very old version of the toolchain:
>
> ```
> WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
> Expected to see version: [...]

Ok Ill try that, thanks

Jimmy106-e, 2024-03-22:
> You seem to be using what seems to be a very old version of the toolchain:
>
> ```
> WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
> Expected to see version: [...]

Just wondering, I installed the toolchain through [here](https://gitdemo.readthedocs.io/en/latest/macos-setup.html) and that is the 'latest' version. Is there another link for the updated version of the toolchain?

igrr, 2024-03-22:
From that page you linked:

> © Copyright 2016 - 2017, Espressif. Revision 4de97554.

This is someone's fork of ESP-IDF from 2017, not our official documentation. The up-to-date instructions are at the link from my previous message: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/index.html

Jimmy106-e, 2024-03-22:
> You seem to be using what seems to be a very old version of the toolchain:
>
> ```
> WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-61-gab8375a
> Expected to see version: [...]

Hey! Just fixed it, thanks!

closed 2024-03-22 by Jimmy106-e