espressif/esp-idf #13454: Multiple 'Type' Errors

Jimmy106-e, 2024-03-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Operating System used.

macOS

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Should have flashed firmware to Esp32 Cam

### What is the actual behavior?

Instead it gave me multiple Type errors

### Steps to reproduce.

1. Follow [this](https://www.studiopieters.nl/esp32-homekit-camera/) guide.
2. Start to flash
3. Errors show up

### Build or installation Logs.

```plain
Toolchain path: /Users/DGB/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /Users/DGB/esp32/esp-idf/requirements.txt are satisfied.
CC build/app_trace/app_trace.o
CC build/app_trace/app_trace_util.o
CC build/app_trace/host_file_io.o
CC build/app_trace/gcov/gcov_rtio.o
AR build/app_trace/libapp_trace.a
CC build/app_update/esp_ota_ops.o
AR build/app_update/libapp_update.a
CXX build/asio/asio/asio/src/asio.o
AR build/asio/libasio.a
AR build/aws_iot/libaws_iot.a
CC build/bootloader_support/src/bootloader_clock.o
CC build/bootloader_support/src/bootloader_common.o
CC build/bootloader_support/src/bootloader_flash.o
CC build/bootloader_support/src/bootloader_random.o
CC build/bootloader_support/src/bootloader_sha.o
CC build/bootloader_support/src/bootloader_utility.o
CC build/bootloader_support/src/efuse.o
CC build/bootloader_support/src/esp_image_format.o
CC build/bootloader_support/src/flash_encrypt.o
CC build/bootloader_support/src/flash_partitions.o
CC build/bootloader_support/src/flash_qio_mode.o
CC build/bootloader_support/src/secure_boot.o
CC build/bootloader_support/src/secure_boot_signatures.o
AR build/bootloader_support/libbootloader_support.a
CC build/bt/bt.o
AR build/bt/libbt.a
CC build/coap/libcoap/src/address.o
[... 218 lines cut ...]
     return text_to_stringv(1, (const byte*[]){ data }, (size_t[]){ size });
            ^
/Users/DGB/Desktop/esp32-homekit-camera/components/homekit/src/debug.c:13:12: warning: return makes pointer from integer without a cast [-Wint-conversion]
/Users/DGB/Desktop/esp32-homekit-camera/components/homekit/src/debug.c: In function 'binary_to_string':
/Users/DGB/Desktop/esp32-homekit-camera/components/homekit/src/debug.c:18:12: error: implicit declaration of function 'binary_to_stringv' [-Werror=implicit-function-declaration]
     return binary_to_stringv(1, (const byte*[]){ data }, (size_t[]){ size });
            ^
/Users/DGB/Desktop/esp32-homekit-camera/components/homekit/src/debug.c:18:12: warning: return makes pointer from integer without a cast [-Wint-conversion]
/Users/DGB/Desktop/esp32-homekit-camera/components/homekit/src/debug.c: At top level:
/Users/DGB/Desktop/esp32-homekit-camera/components/homekit/src/debug.c:22:24: error: unknown type name 'uint8_t'
 bool is_human_readable(uint8_t n, const byte **datas, size_t *sizes) {
                        ^
/Users/DGB/Desktop/esp32-homekit-camera/components/homekit/src/debug.c:41:23: error: unknown type name 'uint8_t'
 char *text_to_stringv(uint8_t n, const byte **datas, size_t *sizes) {
                       ^
/Users/DGB/Desktop/esp32-homekit-camera/components/homekit/src/debug.c:79:25: error: unknown type name 'uint8_t'
 char *binary_to_stringv(uint8_t n, const byte **datas, size_t *sizes) {
                         ^
/Users/DGB/Desktop/esp32-homekit-camera/components/homekit/src/debug.c:109:23: error: unknown type name 'uint8_'
 char *data_to_stringv(uint8_t n, const byte **datas, size_t *sizes) {
                       ^
cc1: some warnings being treated as errors
make[1]: *** [src/debug.o] Error 1
make: *** [component-homekit-build] Error 2
```

igrr, 2024-03-22:
```
In file included from /Users/DGB/Desktop/esp32-homekit-camera/components/homekit/src/debug.c:4:0:
/Users/DGB/Desktop/esp32-homekit-camera/components/homekit/src/debug.h:33:23: error: unknown type name 'uint8_t'
 char *data_to_stringv(uint8_t n, const byte **datas, size_t *sizes);
```

The error is reported in your project — esp32-homekit-camera, not in ESP-IDF. So I'm afraid you'll have to fix it yourself.

---

(Note, continuing on https://github.com/espressif/esp-idf/issues/13453#issuecomment-2015370756 — you are still using a very old version of ESP-IDF, probably v3.3.2 or older. You have not "fixed" that part of the problem. IDF v3.3.x is EOL since 2022.)

Jimmy106-e, 2024-03-22:
> ```
> In file included from /Users/DGB/Desktop/esp32-homekit-camera/components/homekit/src/debug.c:4:0:
> /Users/DGB/Desktop/esp32-homekit-camera/components/homekit/src/debug.h:33:23: error: unknown [...]

Ok sorry to post here.

closed 2024-03-22 by Jimmy106-e