espressif/esp-idf #13455: a7 cannot be used in asm here

MATTYGILO, 2024-03-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am building for ESP32S3

I am getting this error:
```
lib/l3mdct.c:136:1: error: a7 cannot be used in asm here
 }
 ^
lib/l3subband.c: In function 'shine_window_filter_subband':
lib/l3subband.c:106:1: error: a7 cannot be used in asm here
 }
 ^
``` 

Here is all my code:
[main.zip](https://github.com/espressif/esp-idf/files/14728600/main.zip)

Dazza0, 2024-03-23:
Xtensa ISA uses A7 as the frame pointer. Usage of frame pointers are compiled into the code whenever `O0` optimization is specified. Could you try either...

- Compiling your code with the `-fomit-frame-pointer` option
- Using one of the compiler optimizations (e.g., `Og`, `Os`, `O2`)

KonstantinKondrashov, 2024-03-25:
Hi @MATTYGILO!
The lib in your project uses asms. I see that some asms directly use A7(and A8). Probably you may need to update it to use `%0` technique instead.

```asm
#define muladd(hi, lo, x, y) \
 ({  \
    asm ( \
    		"mulsh a7, %2, %1\n\t" \
    		"add %0, %0, a7\n\t" \
    		: "+r" (hi) : "r" (x), "r" (y) \
    		: "a7");\
})
```

MATTYGILO, 2024-03-25:
Ok so the code I was basing off said to compile with:
```
target_compile_options(usermod_pymp3 INTERFACE
    -O3
    -ffast-math
)
```

closed 2024-03-25 by MATTYGILO