espressif/esp-idf #13457: WiFi Enhanced Open in soft AP mode

kevinhikaruevans, 2024-03-23:
### Is your feature request related to a problem?

Open wifi networks can be insecure to users, since an attacker can eavesdrop on wifi packets in the air. I have an application where I don't want to go through the hassle of distributing a PSK but cannot have eavesdroppers. 

### Describe the solution you'd like.

I'd like support for OWE/Enhanced Open when broadcasting a soft AP.  It already exists for STA mode. 

### Describe alternatives you've considered.

- WPA - I cannot easily distribute the PSK, so this is infeasible 
- Open wifi and HTTPS for this purpose, but I don't want to have to distribute keys and have users install them.

nishanth-radja, 2024-04-04:
@kevinhikaruevans will wps registrar with PBC help for this use case ?

kevinhikaruevans, 2024-04-04:
I don't think so unfortunately. I'd really like to have an open network for this use case.

kevinhikaruevans, 2024-10-11:
Hello, I just wanted to see if this task was on the roadmap and when this might get added into ESP-IDF. Thank you!

nachiketkukade, 2024-11-11:
@kevinhikaruevans , this is a big feature and it's currently not on the Wi-Fi feature roadmap. The ideal use case here would be to have an AP with Open + OWE mixed mode, so that legacy devices (most common case) can fall back to Open mode. But this requires dual Beaconing which is a big requirement for a SoftAP.