espressif/esp-idf #13458: Change version from 5.2.1 to master - can't compile now

MaciekMr, 2024-03-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1 / master

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

1. Installing version v5.2.1
2. Compile projects
3. Delete folders .espressif and 5.2.1
4. Install new version of esp-idf - master (latest)
5. Reinstall VSCode
6. Trying to reinstall projects
7. Compiler is looking for 5.2.1 folder instead on master
CMake Error at CMakeLists.txt:7 (include):
  include could not find requested file:

    E:\tools\ESP32_dev\v5.2.1\esp-idf/tools/cmake/project.cmake

CMakeList.txt:
.....
include($ENV{IDF_PATH}/tools/cmake/project.cmake)

Removing /build folders - no effect

### What is the actual behavior?

Compiler should take a master folder

### Steps to reproduce.

1. Installing version v5.2.1
2. Compile projects
3. Delete folders .espressif and 5.2.1
4. Install new version of esp-idf - master (latest)

### Build or installation Logs.

```plain
CMake Error at CMakeLists.txt:7 (include):
  include could not find requested file:

    E:\tools\ESP32_dev\v5.2.1\esp-idf/tools/cmake/project.cmake

-- The C compiler identification is GNU 13.1.0
-- The CXX compiler identification is GNU 13.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: E:/tools/Strawberry/c/bin/gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: E:/tools/Strawberry/c/bin/c++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring incomplete, errors occurred!
See also "E:/Projects/ESP32/test/build/CMakeFiles/CMakeOutput.log".

 *  The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B e:\Projects\ESP32\test\build -S e:\Projects\ESP32\test" terminated with exit code: 1.
```

### More Information.

Where I can change IDF_PATH?

SolidStateLEDLighting, 2024-03-24:
Here is what I would do.

Make sure that through ESP-IDF: Configure ESP-IDF Extension command (Use Existing Setup) that the correct IDF is made active.

Create a fresh "Hello World" application near this project (use ESP-IDF:New Project command).   Get that to compile.

Then copy all the important IDF and Tool pathways from the Hello Project over to your target project.

The items you need to update will be inside .vscode/c_cpp_properties.json and .vscode/settings.json

MaciekMr, 2024-03-24:
I have update to latest master today and it started working.
Ticket to close

closed 2024-03-27 by Alvin1Zhang