espressif/esp-idf #13459: esp_http_client_set_url() parse failure

espradio, 2024-03-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

http://croonerradio_classicgold.ice.infomaniak.ch/croonerradio-classicgold-midfi.mp3 URL fails on esp_http_client_set_url()
with ESP_ERR_INVALID_ARG. However, it is a valid URL.

espradio, 2024-03-24:
Similar issue with https://amorfm_192.streampartner.nl/stream

hmalpani, 2024-03-26:
Hello @espradio 
http_parser does not support **"_"** i.e **underscore** character in host name part of the URL by default. This is the reason you are getting the error while calling `esp_http_client_set_url()` API for setting the URL. You can enable by adding the following in project's top level CMakeList.txt:
```
idf_component_get_property(http_parser http_parser COMPONENT_LIB)
target_compile_definitions(${http_parser} PRIVATE "-DHTTP_PARSER_STRICT=0")
```

Hope it helps.

hmalpani, 2024-04-01:
Closing this issue. Please feel free to reopen if the issue persists.

closed 2024-04-01 by hmalpani

espradio, 2024-04-13:
@hmalpani Thanks it works.