espressif/esp-idf #13464: How do I change DNS to STA access Amazon Cloud using esp32 c3

SeanYang-7, 2024-03-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Because sometimes I can't connect to the Amazon cloud, I need to change the dns to Google's

Alvin1Zhang, 2024-03-25:
Thanks for reporting, would you please help provide more details as per [the issue template](https://github.com/espressif/esp-idf/issues/new/choose)? Thanks.

SeanYang-7, 2024-03-25:
> Thanks for reporting, would you please help provide more details as per [the issue template](https://github.com/espressif/esp-idf/issues/new/choose)? Thanks.
Thank you very much for your reply, I want to set up the dns connection of my device to find Amazon Cloud after connecting to the router, but I don't know how to do that

SeanYang-7, 2024-03-25:
[image]

closed 2024-03-25 by SeanYang-7