espressif/esp-idf #13467: Heap task tracking is broken in v5.2.1

yyjdelete, 2024-03-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1 N16R8

### Power Supply used.

USB

### What is the expected behavior?

It should work, as it does in 5.1 branch.

### What is the actual behavior?

It will report an invalid task handle: `0x580e`

And debug show that for the block owns invalid task handle, in `heap_task_info.c`->`heap_caps_get_per_task_info()` the `p` is always equals to `reg`, which should be malloc in `heap_caps_init.c`
https://github.com/espressif/esp-idf/blob/v5.2.1/components/heap/heap_task_info.c#L27-L71
https://github.com/espressif/esp-idf/blob/v5.2.1/components/heap/heap_caps_init.c#L128

### Steps to reproduce.

Just build and run `examples/system/heap_task_tracking`

KonstantinKondrashov, 2024-03-25:
Hi @yyjdelete! Thanks for the report. I confirm that there is an issue. We are working to fix it.

SoucheSouche, 2024-04-09:
hi @yyjdelete,
I found and fixed the problem.
Thanks again for reporting it.

closed 2024-04-17 by espressif-bot